eclipse-threadx/netxduo #4: TCP Fast retransmission due to faulty DUPACK counting

MathieuBordere, 2020-05-27:
https://github.com/azure-rtos/netxduo/blob/35a977685463af4b756d5396b9bcb73f33eae9ff/common/src/nx_tcp_socket_state_ack_check.c#L232-L235

If I'm not mistaken the `socket_ptr -> nx_tcp_socket_duplicated_ack_received` is increased on segments that contain data, from my understanding a 'real' DUPACK is a pure ACK packet that does not contain data, `socket_ptr -> nx_tcp_socket_duplicated_ack_received` should therefore only be increased if the segment contains no data. We have observed this behaviour where NetX initiates a fast retransmit without the receiver explicitly signaling packet loss by sending DUPACKs.

yuxin-azrtos, 2020-05-28:
@MathieuBordere Thank you for sharing the finding with us.  We reviewed the code and I am able to confirm that is an issue in the TCP logic. I have forwarded this information to the NetX team for further analysis the issue and fix it.  Feel free to re-open this ticket if you have additional information.

closed 2020-05-28 by yuxin-azrtos

MathieuBordere, 2020-05-29:
Is there a way to keep me posted when a fix is issued, or should I seek support through my employer who owns a ThreadX & NetX license?