eclipse-threadx/netxduo #5: broker-less pubsub?

edbek, 2020-06-28:
Are there any plans in the future to add the possibility of broker-less pubsub technology to netxduo, for example, nng (https://github.com/nanomsg/nng)?

edbek, 2020-07-15:
Such a product as NNG is no more. Therefore, if the NNG appears on your network stack, then this will be a serious advance.

yuxin-azrtos, 2020-07-17:
edbek, this is an interesting idea.  I did some search but couldn't find the spec on the serverless pubsub.  Can you point me the right direction? Thanks!

edbek, 2020-07-18:
I can advise you to refer to the documentation for the ZeroMQ project https://zeromq.org/, since the NNG project itself was inspired by ZeroMQ.. The ZeroMQ project itself is very large for small embedded systems, but NNG is suitable for this.
Hope this helps you. Thank.

yuxin-azrtos, 2020-07-31:
Thanks for the suggestion. We don't have plans to support ZeroMQ at this point.

closed 2020-07-31 by yuxin-azrtos