eclipse-threadx/netxduo #6: Additional toolchain

cyliangtw, 2020-07-20:
According to the last commit, it mentioned "`updated to 6.0.1 and added additional processors/toolchains`".
It's great in "`\threadx\ports\`", it really supports multiple toolchains, like as iar, kiel.
However in "`\netxduo\ports\cortex_m4`", it's still gnu toolchain only.
When will push the code for  iar or kiel toolchain ?

yuxin-azrtos, 2020-07-20:
Cyliangtw, additional processor/toolchain support will be added to the rest of the Azure RTOS components in the coming weeks.

cyliangtw, 2020-07-21:
@yuxin-azrtos , thanks of your reply. 
We are porting the connectivity based on nxd in Keil and it's better to apply your next commit.

yuxin-azrtos, 2020-08-08:
I am closing this issue as our engineering team is working on the additional ports.  These ports will be released as they become ready.

closed 2020-08-08 by yuxin-azrtos