eclipse-threadx/netxduo #7: Integer-to-pointer cast in _nx_ip_thread_entry

gamozolabs, 2020-08-29:
Heya,

In `_nx_ip_thread_entry` there is a `ULONG` (which may be an `unsigned int` on 64-bit Linux targets) which gets promoted directly to a `NX_IP*`.

The compiler gets upset about this too if `-Werror` is used.

```
In file included from /home/pleb/netxduo_fuzzing/git/netxduo/common/src/nx_ip_thread_entry.c:28:
/home/pleb/netxduo_fuzzing/git/netxduo/common/src/nx_ip_thread_entry.c: In function '_nx_ip_thread_entry':
../git/netxduo/common/inc/nx_api.h:131:71: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
  131 |                                                                 (a) = (b *)(c); \
      |                                                                       ^
/home/pleb/netxduo_fuzzing/git/netxduo/common/src/nx_ip_thread_entry.c:135:5: note: in expansion of macro 'NX_THREAD_EXTENSION_PTR_GET'
  135 |     NX_THREAD_EXTENSION_PTR_GET(ip_ptr, NX_IP, ip_ptr_value)
```

-B

gamozolabs, 2020-08-29:
Here's the whole list of pointer and `ULONG` mismatches in the codebase.

```
/home/pleb/netxduo_fuzzing/git/netxduo/common/src/nx_ip_thread_entry.c: In function '_nx_ip_thread_entry':
../git/netxduo/common/inc/nx_api.h:131:71: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  131 |                                                                 (a) = (b *)(c); \
      |                                                                       ^
/home/pleb/netxduo_fuzzing/git/netxduo/common/src/nx_ip_thread_entry.c:135:5: note: in expansion of macro 'NX_THREAD_EXTENSION_PTR_GET'
  135 |     NX_THREAD_EXTENSION_PTR_GET(ip_ptr, NX_IP, ip_ptr_value)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pleb/netxduo_fuzzing/git/netxduo/common/src/nx_ip_fast_periodic_timer_entry.c:28:
/home/pleb/netxduo_fuzzing/git/netxduo/common/src/nx_ip_fast_periodic_timer_entry.c: In function '_nx_ip_fast_periodic_timer_entry':
../git/netxduo/common/inc/nx_api.h:141:71: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  141 |                                                                 (a) = (b *)(c); \
      |                                                                       ^
/home/pleb/netxduo_fuzzing/git/netxduo/common/src/nx_ip_fast_periodic_timer_entry.c:81:5: note: in expansion of macro 'NX_TIMER_EXTENSION_PTR_GET'
   81 |     NX_TIMER_EXTENSION_PTR_GET(ip_ptr, NX_IP, ip_address)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pleb/netxduo_fuzzing/git/netxduo/common/src/nx_ip_periodic_timer_entry.c:28:
/home/pleb/netxduo_fuzzing/git/netxduo/common/src/nx_ip_periodic_timer_entry.c: In function '_nx_ip_periodic_timer_entry':
../git/netxduo/common/inc/nx_api.h:141:71: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  141 |                                                                 (a) = (b *)(c); \
      |                                                                       ^
/home/pleb/netxduo_fuzzing/git/netxduo/common/src/nx_ip_periodic_timer_entry.c:81:5: note: in expansion of macro 'NX_TIMER_EXTENSION_PTR_GET'
   81 |     NX_TIMER_EXTENSION_PTR_GET(ip_ptr, NX_IP, ip_address)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pleb/netxduo_fuzzing/git/threadx/common/src/tx_misra.c: In function '_tx_misra_pointer_to_ulong_convert':
/home/pleb/netxduo_fuzzing/git/threadx/common/src/tx_misra.c:109:12: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  109 |     return((ULONG) ptr);
      |            ^
/home/pleb/netxduo_fuzzing/git/threadx/common/src/tx_misra.c: In function '_tx_misra_ulong_to_pointer_convert':
/home/pleb/netxduo_fuzzing/git/threadx/common/src/tx_misra.c:168:12: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  168 |     return((VOID *) input);
      |            ^
/home/pleb/netxduo_fuzzing/git/threadx/common/src/tx_misra.c: In function '_tx_misra_ulong_to_thread_pointer_convert':
/home/pleb/netxduo_fuzzing/git/threadx/common/src/tx_misra.c:690:27: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  690 |     return((TX_THREAD *) ((VOID *) value));
      |                           ^
In file included from /home/pleb/netxduo_fuzzing/git/netxduo/addons/dhcp/nxd_dhcp_server.c:34:
/home/pleb/netxduo_fuzzing/git/netxduo/addons/dhcp/nxd_dhcp_server.c: In function '_nx_dhcp_fast_periodic_timer_entry':
../git/netxduo/common/inc/nx_api.h:141:71: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  141 |                                                                 (a) = (b *)(c); \
      |                                                                       ^
/home/pleb/netxduo_fuzzing/git/netxduo/addons/dhcp/nxd_dhcp_server.c:485:5: note: in expansion of macro 'NX_TIMER_EXTENSION_PTR_GET'
  485 |     NX_TIMER_EXTENSION_PTR_GET(dhcp_ptr, NX_DHCP_SERVER, info)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pleb/netxduo_fuzzing/git/netxduo/addons/dhcp/nxd_dhcp_server.c: In function '_nx_dhcp_slow_periodic_timer_entry':
../git/netxduo/common/inc/nx_api.h:141:71: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  141 |                                                                 (a) = (b *)(c); \
      |                                                                       ^
/home/pleb/netxduo_fuzzing/git/netxduo/addons/dhcp/nxd_dhcp_server.c:540:5: note: in expansion of macro 'NX_TIMER_EXTENSION_PTR_GET'
  540 |     NX_TIMER_EXTENSION_PTR_GET(dhcp_ptr, NX_DHCP_SERVER, info)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pleb/netxduo_fuzzing/git/netxduo/addons/dhcp/nxd_dhcp_server.c: In function '_nx_dhcp_server_thread_entry':
../git/netxduo/common/inc/nx_api.h:131:71: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  131 |                                                                 (a) = (b *)(c); \
      |                                                                       ^
/home/pleb/netxduo_fuzzing/git/netxduo/addons/dhcp/nxd_dhcp_server.c:1515:5: note: in expansion of macro 'NX_THREAD_EXTENSION_PTR_GET'
 1515 |     NX_THREAD_EXTENSION_PTR_GET(dhcp_ptr, NX_DHCP_SERVER, info)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pleb/netxduo_fuzzing/git/netxduo/addons/dhcp/nxd_dhcpv6_server.c:35:
[... cut ...]
../git/netxduo/common/inc/nx_api.h:131:71: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  131 |                                                                 (a) = (b *)(c); \
      |                                                                       ^
/home/pleb/netxduo_fuzzing/git/netxduo/addons/dhcp/nxd_dhcpv6_client.c:10066:5: note: in expansion of macro 'NX_THREAD_EXTENSION_PTR_GET'
10066 |     NX_THREAD_EXTENSION_PTR_GET(dhcpv6_ptr, NX_DHCPV6, info)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
```

goldscott, 2020-09-01:
We haven’t released 64-bit port files on github yet. So for now you'll need to specify -m32 for both compiler and link flags.

yuxin-azrtos, 2020-09-03:
This issue will be resolved once we release 64-bit support.  Closing the issue now. Feel free to re-open if you have additional questions.

closed 2020-09-03 by yuxin-azrtos