eclipse-threadx/netxduo #12: Incorrect type conversion to double

ryanwinter, 2020-12-01:
The following line contains an unnecessary cast from a uint32_t to a double, and back to a uint32_t.

https://github.com/azure-rtos/netxduo/blob/7413f09e70875ad465f5658fdb5a59902778bf50/addons/azure_iot/azure_iot_security_module/iot-security-module-core/src/core.c#L126

This is generating a gcc compiler warning (for the Renesas RX GCC compiler):
```
C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/azure_iot_security_module/iot-security-module-core/src/core.c:126:31: error: conversion from 'uint32_t' {aka 'long unsigned int'} to 'double' may change value [-Werror=conversion]
             double interval = priority_collectors_get_interval(prioritized_collectors);
```

ryanwinter, 2020-12-01:
@illandsb

illandsb, 2021-01-29:
The fix is ready on DevOps repo and planned to be published on March release

yuxin-azrtos, 2021-02-05:
Close this issue as it will be released in March.

closed 2021-02-05 by yuxin-azrtos