eclipse-threadx/netxduo #13: Invalid cast from "unsigned int*" to "size_t*" for Renesas RX GCC compiler in middleware

ryanwinter, 2020-12-01:
Integration with the new RX65N RSK, the RX GCC compiler has defined size_t as 8 bytes. This causes a number of compiler errors in places where an **unsigned int*** is passed into a function expecting a **size_t***.

Compiler: 
https://gcc-renesas.com/rx-download-toolchains/

Error:
```
FAILED: lib/netxduo/CMakeFiles/netxduo.dir/addons/azure_iot/nx_azure_iot_hub_client.c.obj
"C:\Users\ryanw\AppData\Roaming\GCC for Renesas RX 8.3.0.202004-GNURX-ELF\rx-elf\rx-elf\bin\rx-elf-gcc.exe" -DASC_SERIALIZER_USE_CUSTOM_ALLOCATOR -DLOG_LEVEL=0 -DNX_INCLUDE_USER_DEFINE_FILE -DTX_INCLUDE_USER_DEFINE_FILE -IC:/projects/iot/getting-started/core/lib/netxduo/ports/rx65n/gnu/inc -IC:/projects/iot/getting-started/core/lib/netxduo/common/inc -IC:/projects/iot/getting-started/core/lib/netxduo/addons/auto_ip -IC:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot -IC:/projects/iot/getting-started/core/lib/netxduo/addons/BSD -IC:/projects/iot/getting-started/core/lib/netxduo/addons/cloud -IC:/projects/iot/getting-started/core/lib/netxduo/addons/dhcp -IC:/projects/iot/getting-started/core/lib/netxduo/addons/dns -IC:/projects/iot/getting-started/core/lib/netxduo/addons/ftp -IC:/projects/iot/getting-started/core/lib/netxduo/addons/http -IC:/projects/iot/getting-started/core/lib/netxduo/addons/mdns -IC:/projects/iot/getting-started/core/lib/netxduo/addons/mqtt -IC:/projects/iot/getting-started/core/lib/netxduo/addons/nat -IC:/projects/iot/getting-started/core/lib/netxduo/addons/pop3 -IC:/projects/iot/getting-started/core/lib/netxduo/addons/ppp -IC:/projects/iot/getting-started/core/lib/netxduo/addons/pppoe -IC:/projects/iot/getting-started/core/lib/netxduo/addons/smtp -IC:/projects/iot/getting-started/core/lib/netxduo/addons/snmp -IC:/projects/iot/getting-started/core/lib/netxduo/addons/sntp -IC:/projects/iot/getting-started/core/lib/netxduo/addons/telnet -IC:/projects/iot/getting-started/core/lib/netxduo/addons/tftp -IC:/projects/iot/getting-started/core/lib/netxduo/addons/web -IC:/projects/iot/getting-started/core/lib/netxduo/crypto_libraries/inc -IC:/projects/iot/getting-started/core/lib/netxduo/nx_secure/inc -IC:/projects/iot/getting-started/core/lib/netxduo/nx_secure/ports -Ilib/netxduo/custom_inc -IC:/projects/iot/getting-started/core/lib/threadx/ports/rx65n/gnu/inc -IC:/projects/iot/getting-started/core/lib/threadx/common/inc -Ilib/threadx/custom_inc -IC:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/azure-sdk-for-c/sdk/inc -IC:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/azure-sdk-for-c/sdk/tests/core/inc -IC:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/azure_iot_security_module -IC:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/azure_iot_security_module/inc -IC:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/azure_iot_security_module/iot-security-module-core/inc -mcpu=rx64m -misa=v2 -mlittle-endian-data  --specs=nano.specs -ffunction-sections -fdata-sections -fno-strict-aliasing -fno-builtin -fshort-enums -Wall -Wextra -Wuninitialized -Wshadow -Wdouble-promotion -Werror -Wno-unused-parameter -O0 -g3 -std=gnu99 -MD -MT lib/netxduo/CMakeFiles/netxduo.dir/addons/azure_iot/nx_azure_iot_hub_client.c.obj -MF lib\netxduo\CMakeFiles\netxduo.dir\addons\azure_iot\nx_azure_iot_hub_client.c.obj.d -o lib/netxduo/CMakeFiles/netxduo.dir/addons/azure_iot/nx_azure_iot_hub_client.c.obj -c C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.c
C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.c: In function 'nx_azure_iot_hub_client_connect':
C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.c:305:86: error: passing argument 4 of 'az_iot_hub_client_get_client_id' from incompatible pointer type [-Werror=incompatible-pointer-types]
                                                   (CHAR *)buffer_ptr, buffer_length, &buffer_length);
                                                                                      ^~~~~~~~~~~~~~
In file included from C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.h:30,
                 from C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.c:14:
C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/azure-sdk-for-c/sdk/inc/azure/iot/az_iot_hub_client.h:153:13: note: expected 'size_t *' {aka 'long unsigned int *'} but argument is of type 'UINT *' {aka 'unsigned int *'}
     size_t* out_mqtt_client_id_length);
     ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.c:325:86: error: passing argument 4 of 'az_iot_hub_client_get_user_name' from incompatible pointer type [-Werror=incompatible-pointer-types]
                                                   (CHAR *)buffer_ptr, buffer_length, &buffer_length);
                                                                                      ^~~~~~~~~~~~~~
In file included from C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.h:30,
                 from C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.c:14:
C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/azure-sdk-for-c/sdk/inc/azure/iot/az_iot_hub_client.h:131:13: note: expected 'size_t *' {aka 'long unsigned int *'} but argument is of type 'UINT *' {aka 'unsigned int *'}
     size_t* out_mqtt_user_name_length);
     ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.c: In function 'nx_azure_iot_hub_client_telemetry_message_create':
C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.c:789:79: error: passing argument 5 of 'az_iot_hub_client_telemetry_get_publish_topic' from incompatible pointer type [-Werror=incompatible-pointer-types]
                                                                 topic_length, &topic_length);
                                                                               ^~~~~~~~~~~~~
In file included from C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.h:30,
                 from C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.c:14:
C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/azure-sdk-for-c/sdk/inc/azure/iot/az_iot_hub_client.h:244:13: note: expected 'size_t *' {aka 'long unsigned int *'} but argument is of type 'UINT *' {aka 'unsigned int *'}
     size_t* out_mqtt_topic_length);
     ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.c: In function 'nx_azure_iot_hub_client_device_twin_reported_properties_send':
C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.c:1572:79: error: passing argument 5 of 'az_iot_hub_client_twin_patch_get_publish_topic' from incompatible pointer type [-Werror=incompatible-pointer-types]
                                                                  buffer_size, &topic_length);
                                                                               ^~~~~~~~~~~~~
In file included from C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.h:30,
                 from C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.c:14:
C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/azure-sdk-for-c/sdk/inc/azure/iot/az_iot_hub_client.h:464:13: note: expected 'size_t *' {aka 'long unsigned int *'} but argument is of type 'UINT *' {aka 'unsigned int *'}
     size_t* out_mqtt_topic_length);
     ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.c: In function 'nx_azure_iot_hub_client_device_twin_properties_request':
C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_hub_client.c:1744:82: error: passing argument 5 of 'az_iot_hub_client_twin_document_get_publish_topic' from incompatible pointer type [-Werror=incompatible-pointer-types]
                                                                     buffer_size, &topic_length);
[... cut ...]
                 from C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/nx_azure_iot_provisioning_client.c:14:
C:/projects/iot/getting-started/core/lib/netxduo/addons/azure_iot/azure-sdk-for-c/sdk/inc/azure/iot/az_iot_provisioning_client.h:104:13: note: expected 'size_t *' {aka 'long unsigned int *'} but argument is of type 'UINT *' {aka 'unsigned int *'}
     size_t* out_mqtt_user_name_length);
     ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
cc1.exe: all warnings being treated as errors
```

TiejunMS, 2020-12-02:
Can you try to specify `-m32` in the build option? Azure IoT middleware does not support 64 bit yet. It looks like the issue exists in 64 bit environment.

ryanwinter, 2020-12-03:
Hi @TiejunMS,

Yeah this is wierd, the MCU is 32bit, but for some reason they are using 64-bit operations in the default library build.

The -m32 command isn't supported, but poking around there appears to be an alternative build that is 32-bit which it appears I have to specific manually. Let me investigate this and see if I can get it working.

ryanwinter, 2020-12-03:
Investigating further, it appears that an both the "unsigned int" and "long unsigned int" are both 4 bytes. 

The warning is I guess the gcc compiler being overly cautious it casting pointers from one type to another being that on some architectures these types could be different.

TiejunMS, 2020-12-04:
Hi @ryanwinter , could you check with Renesas what is the flags to force 32 bit environment? `-m32` is listed at page 413 of https://gcc-renesas.com/pdf/manuals/gcc.pdf. 

If `long unsigned int` is 4 bytes, could you check the size of `long long`. If it is also 4 bytes, then there is a problem in the NetX crypto which treat it as 8 bytes.

Could you do me a favor to run this piece of code? If sizeof(size_t) is 8 bytes, it makes no sense to convert from `int *` to `size_t *`.
```
#include <stdio.h>

int main()
{
    int a = 0x12345678;
    int x = 0x11111111;
    size_t *b = (size_t *)&a;
    printf("sizeof(int) is %d\n", sizeof(int));
    printf("sizeof(long unsigned int) is %d\n", sizeof(long unsigned int));
    printf("sizeof(size_t) is %d\n", sizeof(size_t));
    printf("&x = %p, x=0x%llx\n", &x, x);
    printf("&a = %p, a=0x%llx\n", &a, a);
    printf("b = %p, *b=0x%llx\n", b, *b);
    return 0;
}
```

ryanwinter, 2020-12-05:
Unfortunately I uninstalled e2studio while I was trying to get something else to work. I'll reinstall and try this again on monday.

The -m32 flag is only for x86_64 architecture, which is why it's not available here. The RX65N is a 32bit processor so the problem here is not that its treating the two types as different in size (size_t, and long unsigned int are both 4 bytes), it that it is flagged an error because of casting one type pointer to another.

ryanwinter, 2020-12-05:
sizeof(int) is 4
sizeof(long unsigned int) is 4
sizeof(size_t) is 4
sizeof(long long) is 8
&x = 0x2724, x=0x272011111111
&a = 0x2720, a=0x272012345678
b = 0x2720, *b=0x272012345678

TiejunMS, 2020-12-07:
Thanks Ryan! Though it is an abnormal behavior of GCC, we will adjust our source code to fix this warning. Will close the issue once it is fixed internally.

ryanwinter, 2020-12-07:
Thanks Tiejun,

I don't think this is abnormal behaviour. I don't think there is a requirement for size_t to be an `unsigned int`, implementations are free to define this how they want. It just happens that in this case it is typedef'd to an `long unsigned int`.

Then is is valid for a compiler to generate a warning when casting from one pointer type to another, as there is no guarantee that the pointer types are compatible. It just worked in the past because size_t on many compilers appears to be of type `unsigned int` so the preprocessor was resolving things before compilation.

TiejunMS, 2020-12-21:
Fixed internally. Will be available in the next release.

closed 2020-12-21 by TiejunMS