eclipse-threadx/netxduo #21: pop3 client port number

ReKawashima, 2021-03-08:
Why does pop3 client use a specific port "4228"?

https://github.com/azure-rtos/netxduo/blob/617d1f0bb79ed1623efdeebabb100e299ef266ce/addons/pop3/nxd_pop3_client.c#L2680

bo-ms, 2021-03-08:
Hi @ReKawashima, thanks for bringing it up, this bug will be fixed in next release. Closing the issue now. Feel free to re-open if you have additional questions.

closed 2021-03-08 by bo-ms

ReKawashima, 2021-03-10:
Hi @bo-ms  -san,
Thank you for response and confirming this bug. I want to know your solution for this bug. Change to "NX_ANY_PORT"?

And, NetX POP3 Client has same bug. Please fix NetX POP3 Client as well.
https://github.com/azure-rtos/netx/blob/490762909bbdb52250c1b491324ba601359ee6ca/addons/pop3/nx_pop3_client.c#L2468