eclipse-threadx/netxduo #22: debugging packets

dvescovi1, 2021-03-22:
I made some progress porting azure RTOS with netxduo to my M4 processor. I have most everything working but having some issues debugging ethernet packets.
The low level Ethernet hardware is setup and it looks like I am sending TX frames (I can see DHCP discover frames in Wireshark coming from my device) correctly but I am having issues with receive. I would like to get more visibility into the rx packets.
I am not sure how to build with packet debugging turned on. How do you do this and am I on the right path?

TiejunMS, 2021-03-23:
If you are able to receive data, then you can compare the data you received with the one captured by wireshark. If not, I suggest you refer to examples and specs of your platform.

dvescovi1, 2021-03-23:
I have already attempted to capture the DHCP offer in Wireshark, unfortunately the offer response is unicast and does not show (I am on a switched network).
But that is not my question. I want to be able to "see" the received data packet information and thus if it get parsed correctly. I see several compile defines that can be switched on to aid debugging ... like NX_DEBUG_PACKET and NX_ENABLE_PACKET_DEBUG_INFO but I do not understand how to used these. I set the #define for these in nx_user.h and rebuilt the app but see little difference. Is it possible to dump packet on terminal? 
I have followed receive to the point the data looks to be queued in the nx_ip_packet_dferred_receive.c but not sure where to look next?

TiejunMS, 2021-03-24:
The option `NX_DEBUG_PACKET ` is used by driver. Here is the reference from ram driver: https://github.com/azure-rtos/netxduo/blob/617d1f0bb79ed1623efdeebabb100e299ef266ce/common/src/nx_ram_network_driver.c#L667  You will need to implement it by yourself.
And option `NX_ENABLE_PACKET_DEBUG_INFO` is used to trace the packet in NetXDuo but not the content.

dvescovi1, 2021-03-24:
little help

closed 2021-03-24 by dvescovi1