eclipse-threadx/netxduo #24: Use bsd sample socket function return TX_MUTEX_ERROR

HarbourXbr, 2021-03-26:
I use the bsd sample , call the socket return –1, i found out the socket call the txe_mutex_get function and the function return TX_MUTEX_ERROR. How to solve this problem

TiejunMS, 2021-03-29:
Could you check whether ThreadX and NetXDuo are rebuilt? See user guide [here](https://docs.microsoft.com/en-us/azure/rtos/netx-duo/netx-duo-bsd/chapter2#building-the-threadx-and-netx-duo-components-of-a-bsd-application).

HarbourXbr, 2021-03-29:
> Could you check whether ThreadX and NetXDuo are rebuilt? See user guide [here](https://docs.microsoft.com/en-us/azure/rtos/netx-duo/netx-duo-bsd/chapter2#building-the-threadx-and-netx-duo-components-of-a-bsd-application).

There is no problem with Ping Sample，ThreadX and NetXDuo are running, i use the bsd sample, the socket function call the txe_mutex_get function to return TX_MUTEX_ERROR, I don't know what caused this problem. When using BSD but not using BSD RAW, i only need to define this macro NX_ENABLE_EXTENDED_NOTIFY_SUPPORT?

TiejunMS, 2021-03-29:
Please confirm if this step is done. 
```
In tx_port.h, set one of the TX_THREAD_EXTENSION macros as follows:

#define TX_THREAD_EXTENSION_3 int bsd_errno
```

HarbourXbr, 2021-03-29:
> Please confirm if this step is done. 
> ```
> In tx_port.h, set one of the TX_THREAD_EXTENSION macros as follows:
> 
> #define TX_THREAD_EXTENSION_3 int bsd_errno
> ```

Yes  i define this macro

TiejunMS, 2021-03-29:
Could you share the following information?
1. Which sample is used? demo_bsd_raw.c, demo_bsd_tcp.c or demo_bsd_udp? 
2. Which board did you use?
3. Which IDE/Compiler did you use?
4. What is the return value of `bsd_initialize`? If it is not zero, could you step into this function and see where it fails?
5. Could you step into txe_mutex_get() and see where it fails?

HarbourXbr, 2021-03-29:
> Could you share the following information?
> 1. Which sample is used? demo_bsd_raw.c, demo_bsd_tcp.c or demo_bsd_udp?
> 2. Which board did you use?
> 3. Which IDE/Compiler did you use?
> 4. What is the [...]
I find the problem, the macro NX_ENABLE_EXTENDED_NOTIFY_SUPPORT need to put nxd_bsd.h , i put it to the nx_usr.h.
Thanks

closed 2021-03-29 by HarbourXbr