eclipse-threadx/netxduo #26: hard fault on _nx_tcp_socket_packet_process

dvescovi1, 2021-04-04:
I have an "almost" fully functioning port to an ARM M4 controller.
I have verified the Threadx demo programs function completely.
The netxduo is very close I believe I have the lowlevel EMAC seetup and mostly working.
I have verified I can send/receive DHCP packets as well as the DNS and SNTP set up.
But I stalls at the point where it actually starts to send/receive TCP based socket packets.
My terminal out puts looks like:
```
 Initializing DNS client

	DNS address: 192.168.1.1

SUCCESS: DNS client initialized

Initializing SNTP client

	SNTP server 0.pool.ntp.org

	SNTP IP address: 171.66.97.126

	SNTP time update: Apr 4, 2021 19:14:1.422 UTC 

SUCCESS: SNTP initialized

Initializing Azure IoT Hub client

	Hub hostname: testhubti.azure-devices.net

	Device id: mydev_2

```
So you can see I am almost there.
The wireshark packets show good handshake for all DHCP/DNS/SNTP packets all the way up to the first TCP packet at which point it makes all the way to the nx_tcp_socket_packet_process.c line 109.

```
VOID  _nx_tcp_socket_packet_process(NX_TCP_SOCKET *socket_ptr, NX_PACKET *packet_ptr)
{

UINT          packet_queued =  NX_FALSE;
NX_TCP_HEADER tcp_header_copy;
VOID          (*urgent_callback)(NX_TCP_SOCKET *socket_ptr);
ULONG         header_length;
ULONG         packet_data_length;
ULONG         packet_sequence;
ULONG         rx_sequence;
ULONG         rx_window;
UINT          outside_of_window;
ULONG         mss = 0;

    /* Add debug information. */
    NX_PACKET_DEBUG(__FILE__, __LINE__, packet_ptr);

    /* Copy the TCP header, since the actual packet can be delivered to
       a waiting socket/thread during this routine and before we are done
       using the header.  */
    /*lint -e{927} -e{826} suppress cast of pointer to pointer, since it is necessary  */
    tcp_header_copy =  *((NX_TCP_HEADER *)packet_ptr -> nx_packet_prepend_ptr);

```
I can set a breakpoint at this last line 
`tcp_header_copy =  *((NX_TCP_HEADER *)packet_ptr -> nx_packet_prepend_ptr); `
and it makes it here but if I do one more step it hard faults.

I am not sure how to interpret the comment or why this fails.

dvescovi1, 2021-04-04:
the last few packets look like:

```
88	16.902087	192.168.1.1	192.168.1.146	DNS	156	Standard query response 0xb7ff A testhubti.azure-devices.net CNAME ihsu-eastus-2.eastus.cloudapp.azure.com A 20.49.109.129
89	16.903396	192.168.1.146	20.49.109.129	TCP	62	54335 → 8883 [SYN] Seq=0 Win=8192 Len=0 MSS=1460
90	16.947130	20.49.109.129	192.168.1.146	TCP	60	8883 → 54335 [SYN, ACK] Seq=0 Ack=1 Win=64240 Len=0 MSS=1440

```

after this I just get a retransmission, I assume because of the failure

```
No.	Time	Source	Destination	Protocol	Length	Info
99	19.950324	20.49.109.129	192.168.1.146	TCP	60	[TCP Retransmission] 8883 → 54335 [SYN, ACK] Seq=0 Ack=1 Win=64240 Len=0 MSS=1440

```

dvescovi1, 2021-04-04:
a bit more information. I was able to correct the hard fault issue by doing this:

```
    /* Copy the TCP header, since the actual packet can be delivered to
       a waiting socket/thread during this routine and before we are done
       using the header.  */
    /*lint -e{927} -e{826} suppress cast of pointer to pointer, since it is necessary  */

//    tcp_header_copy =  *((NX_TCP_HEADER *)packet_ptr -> nx_packet_prepend_ptr);
    memcpy((void *)&tcp_header_copy, (void *)packet_ptr -> nx_packet_prepend_ptr, sizeof(NX_TCP_HEADER));
```
not sure this is a good long term solution. 
Any more insight?

TiejunMS, 2021-04-06:
Could you check the value of `packet_ptr -> nx_packet_prepend_ptr`? It is required to be 4 bytes aligned for IP header. From your last description, it looks like the fix is related to unaligned access.

dvescovi1, 2021-04-06:
Thank you for the help. Yes, it looks like that is the issue.
Even though this is an M4F which I understand supports unaligned access (if the SCB_CCR_UNALIGN_TRP_Msk bit in the SBC->CCR register is clear ... which it is) it is still faulting.
I can see the nx_prepend_ptr is at 0x200037d6 which is indeed in RAM space but not 4 byte aligned.
So what is the answer? do all the buffer allocations need to be fixed up so prepend_ptr end up 4 byte aligned? how do you do that?

goldscott, 2021-04-06:
Hi @dvescovi1 - typically, we use the hardware features of the MAC to insert two bytes of padding between the 14-byte MAC header and the payload. This aligns the payload on a 4-byte boundary.

TiejunMS, 2021-04-14:
Feel free to reopen the issue if you have further questions.

closed 2021-04-14 by TiejunMS