eclipse-threadx/netxduo #28: [Question] "IoT device configuration" IotHub feature for firmware update sample

pesetskyps, 2021-04-13:
There is example of implementation using Preview feature "Device Updates" [here](https://docs.microsoft.com/en-us/azure/iot-hub-device-update/device-update-azure-real-time-operating-system)
Is there any example how to leverage "IoT device configuration" IotHub feature for firmware update?

TiejunMS, 2021-04-14:
Are you looking for sample projects? If so, here is the link: https://github.com/azure-rtos/samples/tree/PublicPreview/ADU

pesetskyps, 2021-04-14:
That is actually example bound to using Azure "Device Updates" service, I am looking for implementation of firmware update using  "IoT device configuration" . The link that you've post is actually the link that is inside the link that I posted in initial question. ADU example are tied to use device update service in Azure.

hihigupt, 2021-04-14:
@pesetskyps just to confirm when you are referring to "IoT device configuration", do you are mean "device twin" ?

pesetskyps, 2021-04-15:
well it levergaes the device twin, but I am talking about this specific tab in Iot Hub
[image]

bo-ms, 2021-04-15:
Hi @pesetskyps  do you mean Device Update for IoT Hub, if so, the tab should be Device updates in IoT Hub, and [here ](https://docs.microsoft.com/en-us/azure/iot-hub-device-update/) is the document about Device Update for IoT Hub.
[image]

pesetskyps, 2021-04-15:
No, Iot device configuration - please take a look at the video. 
https://www.youtube.com/watch?v=pZ9-TLE5RVI&ab_channel=MicrosoftDeveloper

liydu, 2021-04-15:
@pesetskyps, the link you provided is a previous demo we created on top of MXChip board that demonstrates OTA using [Device Twins](https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-devguide-device-twins) in IoT Hub. That was back in the era that Device Update service didn't exist. It should still work but no longer a recommended way to implement the OTA feature using Azure IoT.

So right now, we recommend to use the Azure RTOS and Device Update to build your OTA functionality on the device. You can check more details from: https://aka.ms/azrtos-device-update-preview

bo-ms, 2021-04-17:
Closing this issue now, feel free to open the issue if you have further question.

closed 2021-04-17 by bo-ms