eclipse-threadx/netxduo #29: Question about using a cellular modem with NetXDuo

SL61, 2021-04-21:
Hello, I am planning to use NetXDuo with a cellular modem (in my case, SIM7600 is used). After reading _Chapter 5 - Azure RTOS NetX Duo Network Drivers_, I still don't have a clear clue to port my SIM7600 to NetXDuo. Can you give some suggestions?

TiejunMS, 2021-04-21:
Do you know whether the cellular modem you used support AT mode only or PPP mode? If PPP mode is supported, you will need to develop a PPP driver for data I/O. But the management of cellular initialization or connectivity management is out of the NetX Duo scope.

SL61, 2021-04-21:
> Do you know whether the cellular modem you used support AT mode only or PPP mode? If PPP mode is supported, you will need to develop a PPP driver for data I/O. But the management of cellular initialization or connectivity management is out of the NetX Duo scope.

Both modes are supported and I plan to use AT mode first then use PPP mode later so I will have more flexibility to use components in NetX Duo. Before call NetX Duo api, I have done the initialization (cellular hardware, APN settings and registered on the network). I am thinking I should implement some apis like 

- Cellular_CreateSocket()
- Cellular_SocketConnect()  
- Cellular_SocketSend() 
- Cellular_SocketRecv() 
- Cellular_SocketClose()

and handle URC events.

TiejunMS, 2021-04-21:
To use AT mode, the TCP/IP stack is provided by modem itself. We have a special version of NetXDuo and driver. You can take a look at https://github.com/azure-rtos/samples/releases/download/v6.1_rel/Azure_RTOS_6.1_STM32L4+-DISCO_IAR_Samples_2020_10_10.zip
You will need to port netxduo/common/nx_wifi.c for your modem.

To use PPP mode, you can follow the PPP sample and overwrite ppp_0_serial_byte_output and call nx_ppp_byte_receive when data are received. https://github.com/azure-rtos/netxduo/blob/master/samples/demo_netx_ppp.c

SL61, 2021-04-21:
Thanks, Tiejun, I think it is a good starting point

hwmaier, 2021-04-22:
> ... you will need to develop a PPP driver for data I/O. 

Just a clarification, do you mean by "driver" just the glue code which links PPP and serial port? Because I thought that the PPP network driver itself is part of NetXDuo.

TiejunMS, 2021-04-22:
You are correct.

TiejunMS, 2021-05-06:
Feel free to reopen it if you have more questions.

closed 2021-05-06 by TiejunMS