eclipse-threadx/netxduo #30: Understanding the HTTP server stack.

surajg1608, 2021-05-05:
Hi I wanted to understand how POST and PUT requests are processed in HTTP server.
From the example provided in sample codes [Sample code](https://github.com/azure-rtos/netxduo/blob/master/samples/demo_netxduo_http.c), I understood that it is not mandatory to provide the callback function to process the requests.
If I am correct, I have following concerns/questions assuming that user has not provided callback function to process the requests:
Case 1: PUT request- **General Understanding/Expected behavior**: 
- If file is not present, server shall create the file and write the user provided data.
- If file is present, server shall open the file and append the data.

**Actual results**
- If file is not present, server creates the file and writes data to file: As expected
- If file is present, server returns the error 500 (INTERNAL ERROR)

Case 2: POST request- **General Understanding/Expected behavior**: 
- Server shall create the file and write the data to file.

**Actual results**
- Server checks for existing file, if file is present it returns the content of the file
- If file is not present, server returns the error code 404 (FILE NOT FOUND)

wenhui-xie, 2021-05-06:
You are correct about the behavior you observed. For the PUT method, when the file already exists on the server, NetX HTTP server does not support overwriting it. An alternative solution is to implement it in callback function or use DELETE method to delete the existing file first. We will improve PUT method in the future and let you know when it is ready.

For the POST method, "The actual function performed by the POST method is determined by the server and is usually dependent on the Request-URI." as per [rfc2616](https://tools.ietf.org/html/rfc2616#section-9.5). NetX HTTP server relies on callback function to process the POST request. Without callback function, it is treated similarly as GET request.

TiejunMS, 2021-06-02:
Feel free to reopen it if you have further questions.

closed 2021-06-02 by TiejunMS