eclipse-threadx/netxduo #31: about netxduo ppp question

qq792368480, 2021-05-16:
你好，请原谅我的英文不好表达，请问你们提供的addons里的ppp有实际测试过跟4G模块通信吗？我现在一直卡在LCP阶段，需要自己修改底层吗，看得有点吃力
PPP TX:
7E FF 7D 23 C0 21 7D 21 74 7D 20 7D 28 7D 21 7D |~.}#.!}!t}.}(}!}
24 7D 25 DC BE 20 7E                            |$}%...~

PPP RX:（你们的协议栈已经解包了）
7E FF 03 C0 21 01 00 00 18 02 06 00 00 00 00 03 |~...!...........
04 C0 23 05 06 B4 1C 4F 67 07 02 08 02 C1 28 7E |..#....Og.....(~

PPP TX:
7E FF 7D 23 C0 21 7D 22 7D 20 7D 20 7D 38 7D 22 |~.}#.!}"}.}.}8}"
7D 26 7D 20 7D 20 7D 20 7D 20 7D 23 7D 24 C0 23 |}&}.}.}.}.}#}$.#
7D 25 7D 26 B4 7D 3C 4F 67 7D 27 7D 22 7D 28 7D |}%}&.}<Og}'}"}(}
22 7D 2D C5 7E                                  |"}-.~

PPP RX:
7E FF 03 C0 21 02 74 00 08 01 04 05 DC 6E AA 7E |~...!.t......n.~

还有你们的协议栈有留出打印数据的接口吗，没找到。这样好难调试

bo-ms, 2021-05-16:
@qq792368480  你好，我们PPP有实际测试过。你的应用中应该是PPP + 4G 模块，而我们提供的只是PPP协议栈，用户需要负责4G模块的组网等。关于调试，你可以启用下面两个的宏来查看各个阶段的状态和解码后的数据，如下图所示。另外这里是PPP的中文手册https://docs.microsoft.com/zh-cn/azure/rtos/netx-duo/netx-duo-ppp/chapter1
NX_PPP_DEBUG_LOG_ENABLE 
NX_PPP_DEBUG_LOG_PRINT_ENABLE
[image]

liydu, 2021-05-17:
@qq792368480 你好，我是Azure RTOS的产品经理。请教下你这里用4G走PPP是个什么样的产品？我们最近正在做不少相关的工作，想更详细了解下你这里有关的需求。方便的话可以留个邮箱我们联系到你这里。

qq792368480, 2021-05-17:
> @qq792368480 你好，我们PPP有实际测试过。你的应用中应该是PPP + 4G 模块，而我们提供的只是PPP协议栈，用户需要负责4G模块的组网等。关于调试，你可以启用下面两个的宏来查看各个阶段的状态和解码后的数据，如下图所示。另外这里是PPP的中文手册https://docs.microsoft.com/zh-cn/azure/rtos/netx-duo/netx-duo-ppp/chapter1
> NX_PPP_DEBUG_LOG_ENABLE
> NX_PPP_DEBUG_LOG_PRINT_ENABLE
> [image]

谢谢。我开启调试，打印这些内容，一直停在LCP阶段状态NX_PPP_LCP_PEER_CONFIGURE_REQUEST_ACKED_STATE（4）请问能从以下信息看出是什么原因导致的吗，仿真调试发现ppp_ptr -> nx_ppp_transmit_id != ppp_ptr -> nx_ppp_receive_id。4G模块是已经连接成功才进行ppp协商的
at cmd = ATDT*99#
get len = 11
0D 0A 43 4F 4E 4E 45 43 54 0D 0A                |..CONNECT..
rsp 0
————————————————————————————————————————————————————
Time: 20776, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 1, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Send Packet Length: 10, Packet: c0 21 01 00 00 08 01 04 05 dc 
Time: 21176, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Send Packet Length: 10, Packet: c0 21 01 01 00 08 01 04 05 dc 
Time: 21576, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Send Packet Length: 10, Packet: c0 21 01 02 00 08 01 04 05 dc 
Time: 21748, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 2, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Authenticated, Received Packet Length: 26, Packet: c0 21 01 00 00 18 02 06 00 00 00 00 03 04 c0 23 05 06 ba 6d ef df 07 02 08 02 
Time: 21918, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Send Packet Length: 26, Packet: c0 21 02 00 00 18 02 06 00 00 00 00 03 04 c0 23 05 06 ba 6d ef df 07 02 08 02 
Time: 22276, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Send Packet Length: 10, Packet: c0 21 01 03 00 08 01 04 05 dc 
Time: 22526, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Received Packet Length: 10, Packet: c0 21 02 00 00 08 01 04 05 dc 
Time: 22676, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Send Packet Length: 10, Packet: c0 21 01 04 00 08 01 04 05 dc 
Time: 23048, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Received Packet Length: 10, Packet: c0 21 02 01 00 08 01 04 05 dc 
Time: 23166, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Send Packet Length: 10, Packet: c0 21 01 05 00 08 01 04 05 dc 
Time: 23571, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Received Packet Length: 10, Packet: c0 21 02 02 00 08 01 04 05 dc 
Time: 23690, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Send Packet Length: 10, Packet: c0 21 01 06 00 08 01 04 05 dc 
Time: 24076, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Send Packet Length: 10, Packet: c0 21 01 07 00 08 01 04 05 dc 
Time: 24476, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Send Packet Length: 10, Packet: c0 21 01 08 00 08 01 04 05 dc 
Time: 24675, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Received Packet Length: 26, Packet: c0 21 01 01 00 18 02 06 00 00 00 00 03 04 c0 23 05 06 ba 6d ef df 07 02 08 02 
Time: 24976, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Send Packet Length: 10, Packet: c0 21 01 09 00 08 01 04 05 dc 
Time: 25249, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Received Packet Length: 10, Packet: c0 21 02 03 00 08 01 04 05 dc 
Time: 25376, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Send Packet Length: 10, Packet: c0 21 01 0a 00 08 01 04 05 dc 
Time: 25770, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Received Packet Length: 10, Packet: c0 21 02 04 00 08 01 04 05 dc 
Time: 25890, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Send Packet Length: 10, Packet: c0 21 01 0b 00 08 01 04 05 dc 
Time: 26276, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Send Packet Length: 10, Packet: c0 21 01 0c 00 08 01 04 05 dc 
Time: 26676, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Send Packet Length: 10, Packet: c0 21 01 0d 00 08 01 04 05 dc 
[... cut ...]
Time: 29376, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Send Packet Length: 10, Packet: c0 21 01 13 00 08 01 04 05 dc 
Time: 29651, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 4, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Received Packet Length: 10, Packet: c0 21 02 07 00 08 01 04 05 dc 
Time: 29776, PPP Name: NetX PPP Instance 1, PPP State: 1, PPP LCP State: 1, PPP PAP State: 0, PPP CHAP State: 0, PPP IPCP State: 0, Not Authenticated, Send Packet Length: 10, Packet: c0 21 01 14 00 08 01 04 05 dc

qq792368480, 2021-05-17:
> @qq792368480 你好，我是Azure RTOS的产品经理。请教下你这里用4G走PPP是个什么样的产品？我们最近正在做不少相关的工作，想更详细了解下你这里有关的需求。方便的话可以留个邮箱我们联系到你这里。

用在DTU上面的，当前产品是用lwip实现的，因为netxduo集成了很多插件就自己尝试移植下

bo-ms, 2021-05-18:
> 谢谢。我开启调试，打印这些内容，一直停在LCP阶段状态NX_PPP_LCP_PEER_CONFIGURE_REQUEST_ACKED_STATE（4）请问能从以下信息看出是什么原因导致的吗，仿真调试发现ppp_ptr -> nx_ppp_transmit_id != ppp_ptr -> nx_ppp_receive_id。4G模块是已经连接成功才进行ppp协商的
> at cmd = [...]

ID 是用来匹配请求和响应的。状态4表示对方发送了请求，我们响应了，并且我们发送了请求等待对方响应，但是因为响应的ID不匹配，所以没有完成。你这边测试的PPP server端是仿真的还是实际的？仿真的话你可以调试看看为什么前面3个请求server没有任何响应？并先启动server，然后启动client试试。

qq792368480, 2021-05-18:
> ID 是用来匹配请求和响应的。状态4表示对方发送了请求，我们响应了，并且我们发送了请求等待对方响应，但是因为响应的ID不匹配，所以没有完成。你这边测试的PPP server端是仿真的还是实际的？仿真的话你可以调试看看为什么前面3个请求server没有任何响应？并先启动server，然后启动client试试。

你好，是实际和4G模块进行通信的(EC20，通过串口)，模拟仿真的话采用你们的例程2个PPP接口可以互相通信。对啊我也疑惑前面经过了3个请求包4G模块才响应导致ID不匹配，你们这个PPP包发送间隔是怎么设置的。

bo-ms, 2021-05-18:
默认的重传间隔是4S. nx_ppp.h: NX_PPP_PROTOCOL_TIMEOUT

qq792368480, 2021-05-18:
> 默认的重传间隔是4s. nx_ppp.h: NX_PPP_PROTOCOL_TIMEOUT

谢谢，可以了。是我这边MCU时基设置1ms，
[image]这个没相应改过来

bo-ms, 2021-05-19:
> > 默认的重传间隔是4s. nx_ppp.h: NX_PPP_PROTOCOL_TIMEOUT
> 
> 谢谢，可以了。是我这边MCU时基设置1ms，
> [image]这个没相应改过来

了解，很高兴知道你这边跑通了。请问你能留个联系方式么？想进一步了解一下你这边的应用场景和需求等。

bo-ms, 2021-05-19:
@qq792368480 这个issue我关掉了，如果有任何问题你可以再开新的issue。

closed 2021-05-19 by bo-ms