eclipse-threadx/netxduo #36: Compilation error when compiling NetX Crypto due to conflicting ULONG64 definition

hwmaier, 2021-06-10:
`tx_port.h` for Cortex-M defines a 64-bit type as follows:

    typedef unsigned long long                      ULONG64;

Then in `nx_crypto_huge_number.h` we find this definition:

    #define ULONG64                       unsigned long long

The two definitions result in the gcc compiler to see this after preprocessing:

    typedef unsigned long long                      unsigned long long;

which of course causes compilation errors like these:

```
...\netxduo\crypto_libraries\inc/nx_crypto_huge_number.h:81:39: error: duplicate 'unsigned'
   81 | #define ULONG64                       unsigned long long
...\rm_threadx_port/tx_port.h:90:49: note: in expansion of macro 'ULONG64'
   90 | typedef unsigned long long                      ULONG64;
      |                                                 ^~~~~~~
...\netxduo\crypto_libraries\inc/nx_crypto_huge_number.h:81:48: error: 'long long long' is too long for GCC
   81 | #define ULONG64                       unsigned long long
      |                                                ^~~~
```

Once the conflicting definition of ULONG64  is commented out and the definition for HN_UBASE2 changed to unsigned long long, the offending code compiles.

TiejunMS, 2021-06-11:
Thanks for your feedback! We are working on it.

TiejunMS, 2021-08-04:
Fixed here: https://github.com/azure-rtos/netxduo/blob/d9d7fb05bd6a7a6b50a54c762e1730d8a8054cbb/crypto_libraries/inc/nx_crypto_huge_number.h#L83

closed 2021-08-04 by TiejunMS