eclipse-threadx/netxduo #38: nx_secure_tls_server_state variable should not be removed if using TLS1.3

ReKawashima, 2021-07-02:
In NetX Secure TLS 1.3, I think nx_secure_tls_server_state variable is necessary in TLS Client mode to save the state of server.
But, if disabled server function by NX_SECURE_TLS_SERVER_DISABLED macro to reduce code and data usage, this variable is removed. So, we get build error "NX_SECURE_TLS_SESSION' has no member named 'nx_secure_tls_server_state". 

I think nx_secure_tls_server_state variable should not be removed if using TLS1.3.

I suggest the following change.

Incorrect | **#ifndef   NX_SECURE_TLS_SERVER_DISABLED** <br>       /* The state of the server handshake if   this is a server socket. */ <br>       NX_SECURE_TLS_SERVER_STATE nx_secure_tls_server_state; <br>   #endif <br> (See https://github.com/azure-rtos/netxduo/blob/master/nx_secure/inc/nx_secure_tls.h#L1206)
|:--- | :--- |
Correct | **#if   !defined(NX_SECURE_TLS_SERVER_DISABLED) \|\| (NX_SECURE_TLS_TLS_1_3_ENABLED)** <br>       /* The state of the server handshake if   this is a server socket. */ <br>       NX_SECURE_TLS_SERVER_STATE   nx_secure_tls_server_state; <br>   #endif

TiejunMS, 2021-07-02:
Thanks @ReKawashima ! We will fix it in the next release.

TiejunMS, 2021-08-24:
@ReKawashima , I'm sorry to let you know that the fix is not in the 6.1.8 release. We are working on it right now and will be included in 6.1.9 release.

ReKawashima, 2021-09-10:
@TiejunMS -san, Thank you for information. I'll check the fix after v6.1.9 release. Closing this ticket. Thanks.

closed 2021-09-10 by ReKawashima