eclipse-threadx/netxduo #39: TFTP Client service stops at Block#24

CYG01571, 2021-07-14:
Hi!

I try to transfer a large size data to server by TFTP client service.
I use Synergy S7G2 processor, SSP 2.0.0 and IAR compiler.

I can transfer a small size data(ex. 2048bytes) to server.
But I can't transfer a large size data(Ex. 9MBytes).

The transmit always stops at 24th block.
Please give me a hint to solve.

Best regards.
```
#define WLAN_TFTP_SEND_TIMEOUT (500)
#define WLAN_TFTP_PACKET_SIZE  (512)

static NX_PACKET *my_packet;

extern NX_TFTP_CLIENT g_tftp_client0;

#if 1 //FOR TEST
static uint8_t buf[9*1024*1024+3];
#endif

UINT Tftp_Send(uint8_t *data_addr, uint32_t data_size)
{
	uint32_t status=0;
	UINT server_addr=0;
	uint32_t transfer_cnt=0;
	uint32_t remaining_size=0;
	uint32_t i=0;

#if 1 //FOR TEST
	for(int j=0; j < sizeof(buf); j++){
		buf[j] = j;
	}
	data_addr = buf;
	data_size = sizeof(buf);

#endif
	//calculate trans num
	if( (data_size % WLAN_TFTP_PACKET_SIZE) == 0){
		transfer_cnt = data_size / WLAN_TFTP_PACKET_SIZE;
	}else{
		transfer_cnt = (data_size / WLAN_TFTP_PACKET_SIZE) + 1;
		remaining_size = data_size % WLAN_TFTP_PACKET_SIZE;
	}

	//server address
	server_addr = IP_ADDRESS(192,168,1,3);

	// TFT Open
	status = nx_tftp_client_file_open(&g_tftp_client0, "/test.bin", server_addr, NX_TFTP_OPEN_FOR_WRITE, WLAN_TFTP_SEND_TIMEOUT);

	// Transfer data
	for(i=0; i < transfer_cnt; i++){
		// allocate
		status = nx_tftp_client_packet_allocate(&g_packet_pool0, &my_packet, WLAN_TFTP_SEND_TIMEOUT);

		if(remaining_size==0){
			my_packet -> nx_packet_prepend_ptr = data_addr + WLAN_TFTP_PACKET_SIZE*i;
			// Write pointer update
			my_packet -> nx_packet_length = WLAN_TFTP_PACKET_SIZE;
			my_packet -> nx_packet_append_ptr = my_packet -> nx_packet_prepend_ptr + WLAN_TFTP_PACKET_SIZE;
		}else{
			if(i == transfer_cnt-1){
				my_packet -> nx_packet_prepend_ptr = data_addr + WLAN_TFTP_PACKET_SIZE*i;

				// Write pointer update
				my_packet -> nx_packet_length = remaining_size;
				my_packet -> nx_packet_append_ptr = my_packet -> nx_packet_prepend_ptr + remaining_size;
			}else{
				my_packet -> nx_packet_prepend_ptr = data_addr + WLAN_TFTP_PACKET_SIZE*i;

				// Write pointer update
				my_packet -> nx_packet_length = WLAN_TFTP_PACKET_SIZE;
				my_packet -> nx_packet_append_ptr = my_packet -> nx_packet_prepend_ptr + WLAN_TFTP_PACKET_SIZE;
			}

		}

		// Transfer
		status = nx_tftp_client_file_write(&g_tftp_client0, my_packet, WLAN_TFTP_SEND_TIMEOUT);
	}

	/* close */
	status = nx_tftp_client_file_close(&g_tftp_client0);

	return status;
}

```

bo-ms, 2021-07-14:
Hi @CYG01571 looks the packet points (nx_packet_prepend_ptr, etc) are not set correctly, here is my sample how to send data, could you update your code and test again?

```
VOID tftp_send(UCHAR *data_ptr, UINT data_size)
{
NX_PACKET   *my_packet;
UINT        status;
UINT        packet_length;
ULONG       i;

    /* Loop to send the data.  */
    send_data_length = 0;
    while(send_data_length < data_size)
    {

        /* Allocate a TFTP packet.  */
        status =  nx_tftp_client_packet_allocate(&client_pool, &my_packet, NX_IP_PERIODIC_RATE);
        /* Check status.  */
        if (status)
        {
            printf("ERROR");
            break;
        }

        /* Write ABCs into the packet payload!  */
        if (send_data_length + 512 < data_size)
            packet_length = 512;
        else
            packet_length = data_size - send_data_length;
                                      
        /* Write ABCs into the packet payload!  */
        status = nx_packet_data_append(my_packet, buffer + send_data_length, packet_length, &client_pool, NX_IP_PERIODIC_RATE);

        /* Check status.  */
        if (status)
        {
            printf("ERROR");
            break;
        }

        /* Write this packet to the file via TFTP.  */
        status =  nx_tftp_client_file_write(&client, my_packet, NX_IP_PERIODIC_RATE);

        /* Check status.  */
        if (status)
        {
            printf("ERROR");
            break;
        }

        send_data_length += packet_length;
    }
}
```

CYG01571, 2021-07-15:
Dear bo-ms,

Thank you for your quickly response!
I resolved this issue for your advice.

I could send a large file to TFTP server!
Your advice was very useful because I didn't notice a thing that I should use nx_packet_data_append() API.

Best Regards.

closed 2021-07-15 by CYG01571