eclipse-threadx/netxduo #40: NetX Secure AEAD Cipher checking

ReKawashima, 2021-07-14:
If AEAD Cipher is enabled, AEAD Cipher checking is necessary to detect cipher algorithms other than AES-CCM or AES-GCM. NX_SECURE_AEAD_CIPHER_CHECK macro defines whether the AEAD chcking is used or not.

But, even if NX_SECURE_AEAD_CIPHER_CHECK macro is defined, the AEAD checking is not working (See https://github.com/azure-rtos/netxduo/blob/master/nx_secure/src/nx_secure_tls_record_payload_decrypt.c#L158 ). 
Also, if using cipher algorithms other than AES-CCM or AES-GCM, the communicating counterparty will raise a Bad Record MAC alert.

I think it's a bug in the code. I suggest the following change.

Note: When apply this change, the warning "NX_SECURE_AEAD_CIPHER_CHECK macro redefined" will happened. Please clear the warning.

https://github.com/azure-rtos/netxduo/blob/master/nx_secure/inc/nx_secure_tls.h#L123
<div style='direction:ltr'>

Incorrect | **#ifndef**   NX_SECURE_AEAD_CIPHER_CHECK<br>   #define   NX_SECURE_AEAD_CIPHER_CHECK(a)                  NX_FALSE <br>  #endif /*   NX_SECURE_AEAD_CIPHER_CHECK */
-- | :--
Correct | **#ifdef**   NX_SECURE_AEAD_CIPHER_CHECK &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<-- <br>  #define   NX_SECURE_AEAD_CIPHER_CHECK(a)                  NX_FALSE   <br>#endif /*   NX_SECURE_AEAD_CIPHER_CHECK */

</div>

yanwucai, 2021-07-15:
If you are implementing AEAD ciphers other than AES-CCM or AES-GCM, you can define NX_SECURE_AEAD_CIPHER_CHECK to detect your new algorithms. This macro allows you to extend the conditions to check other algorithms. For example, you can define the macro like:
`#define NX_SECURE_AEAD_CIPHER_CHECK(a) ((a) == YOUR_ALGORITHM_ID)`

ReKawashima, 2021-07-15:
Hi @yanwucai -san,
Thank you for your prompt response.

Sorry. I was misunderstanding.
NX_SECURE_AEAD_CIPHER_CHECK macro defines a additional chcking method, not the use / non-use of the chcking. Right?
If correct, please update sample code (https://github.com/azure-rtos/netxduo/blob/master/nx_secure/inc/nx_secure_user_sample.h#L100) for easy understanding.

Thanks.

TiejunMS, 2021-08-04:
Fixed here: https://github.com/azure-rtos/netxduo/blob/d9d7fb05bd6a7a6b50a54c762e1730d8a8054cbb/nx_secure/inc/nx_secure_user_sample.h#L103

closed 2021-08-04 by TiejunMS