eclipse-threadx/netxduo #42: Socket state race condition in _nx_tcp_socket_send_internal()

hwmaier, 2021-08-03:
I discovered a race condition when `_nx_tcp_socket_send_internal()` is called and pre-empted by the IP thread and the IP thread is changing the socket's state.

We have an application using the `nx_web_http_server` web server component. When running some some tests, we encounter the following situation:

While processing HTTP requests the web server calls `nx_web_http_server_send()`, which calls `nx_tcp_socket_send()` which again calls `nx_tcp_socket_send_internal()`.

Upon entering `nx_tcp_socket_send_internal()`, the state of the socket is checked for possible closure while interrupts are disabled. As soon as interrupts are enabled in line 175, a context switch takes place and the IP thread runs. The IP thread now processes a socket closure on the very same socket we are in the process of sending.

The IP thread changes the state of the socket to `LISTEN` and surrenders CPU.

`nx_web_http_server` thread is scheduled again and execution of `nx_web_http_server_send()` continues. Now the socket is in `LISTEN` state and the `source_ip` and `dest_ip` information is evaluated in line 180 onwards, but this evaluation yields in invalid address information as the socket was closed in the mean time and the address information zeroed.

`nx_web_http_server()` continues execution and aquires the IP mutex (too late now, we already have been preempted) and then calls `_nx_ip_checksum_compute()`.

`_nx_ip_checksum_compute()` will fail when checking the assertions for `src_ip_addr` and `dest_ip_addr`.

```
        NX_ASSERT((src_ip_addr != NX_NULL) && (dest_ip_addr != NX_NULL));
```

and the system will crash.

Here is a TraceX diagram of this scenario:

[image]

I am not sure yet how to best mitigate this race condition. Once a good solution is found, I could test and submit a PR with a fix.

wenhui-xie, 2021-08-11:
Thanks, @hwmaier. We will fix this bug.

hwmaier, 2021-08-11:
Great. BTW, the TraceX tool was essential to pin-point the issue. It is a fantastic tool. 

As I can replicate the bug in my setup, I am happy to beta test the fix.

wenhui-xie, 2021-08-18:
> Great. BTW, the TraceX tool was essential to pin-point the issue. It is a fantastic tool.
> 
> As I can replicate the bug in my setup, I am happy to beta test the fix.

Thanks again @hwmaier. I have fixed the bug. Could you give me your email address and help us to do the beta test? I will email you the fixed source file.

hwmaier, 2021-08-18:
Please email it to hwmaier at gmail dot com.

hwmaier, 2021-08-20:
@wenhui-xie Please email it to hwmaier@gmail.com. Thanks. Happy to test it.

wenhui-xie, 2021-08-20:
@hwmaier I sent you the fixed files, please check your email. Thanks.

yuxin-azrtos, 2021-08-23:
@hwmaier  did you get a chance to evaluate the fix?

hwmaier, 2021-08-23:
@yuxin-azrtos Yes and I have been in contact with @wenhui-xie and confirmed to her the fix to be working. Not sure what your policy is, do you want to close this issue now or wait once the fix has made it into the official github tree.

yuxin-azrtos, 2021-08-23:
@hwmaier we can close it now if you validate the patch fixes your issue.

hwmaier, 2021-08-23:
Closing, provided patch fixes the issue/

closed 2021-08-23 by hwmaier