eclipse-threadx/netxduo #48: Web HTTPS server with TLS1.3 enabled

ReKawashima, 2021-09-10:
Sorry for simple question.
Does AzureRTOS NetX Duo Web HTTPS Server application with TLS1.3 enabled works well?

In my testing, if TLS1.3 enabled with a global macro “NX_SECURE_TLS_ENABLE_TLS_1_3”, an error (the error code returned from nx_secure_tls_process_clienthello_extensions() is NX_SECURE_TLS_UNKOWN_CERT_SIG_ALGORITHM) is occurred. I use RSA SHA256 as signature algorithm in server certificate.

I think this error code "NX_SECURE_TLS_UNKOWN_CERT_SIG_ALGORITHM" means the signature algorithms proposed from client are not matched to one that used in server. This checking are performed in _nx_secure_tls_proc_clienthello_signature_algorithms_extension(). However, if server's local (server) certificate uses RSA, expected signature algorithms were not set, and then the checking result is no matched algorithms (See [here](https://github.com/azure-rtos/netxduo/blob/d9d7fb05bd6a7a6b50a54c762e1730d8a8054cbb/nx_secure/src/nx_secure_tls_process_clienthello_extensions.c#L1156))

tstapko, 2021-09-20:
Thank you for the report. There is a known compilation/configuration issue with TLS 1.3 that we are looking at fixing in the next release.

ReKawashima, 2021-09-22:
@tstapko -san,
Thank you for the update. I'll check the fix after next release. Closing this ticket.
Thanks.

ReKawashima, 2023-01-13:
Hi @tstapko -san, MS team,
Could you please let me know if this issue has been fixed in the latest version (v6.2.0). I've checked the source file ([Link](https://github.com/azure-rtos/netxduo/blob/master/nx_secure/src/nx_secure_tls_process_clienthello_extensions.c#L1178)) and it doesn't seem to be changed/fixed yet.

TiejunMS, 2023-01-13:
@ReKawashima , there is no plan to support RSA-PSS which is required by TLS 1.3 if RSA certificate is used. The only support certificate is ECDSA based for TLS 1.3.

ReKawashima, 2023-01-13:
@TiejunMS -san, I understand. Thanks for your prompt response.

closed 2023-01-13 by ReKawashima