eclipse-threadx/netxduo #49: NetX Secure: How can Secure Client-Initiated Renegotiation be disabled or limited?

hwmaier, 2021-09-14:
How can TLS' *Secure Client-Initiated Renegotiation* be disabled or limited? We tried to set `NX_SECURE_TLS_DISABLE_SECURE_RENEGOTIATION` but this also disables the server initiated renegotation which would weaken security. We only like to the disable client initiated renegotiation or limit its attempt numbers to prevent DoS.

We also tried using `nx_secure_tls_session_renegotiate_callback_set` to implement a limit
  but there seems to be no API hooks to call this function when using the `nx_web_http_server` component.

Could a define like `NX_SECURE_TLS_DISABLE_SECURE_CLIENT_RENEGOTIATION` be introduced ?

  A security audit flagged the *Secure Client-Initiated Renegotiation* as DoS vulnerability (see below) 
  and we like to close this potential security hole. 

  Refer to audit log generated with [ssltest.sh](https://testssl.sh/): 

[image]

tstapko, 2021-09-15:
Thank you for the report! We've determined that this is a feature that should be supported and will prioritize adding it in the next release.

hwmaier, 2021-10-13:
I can see in the latest 6.1.9 release the new `NX_SECURE_TLS_DISABLE_CLIENT_INITIATED_RENEGOTIATION`. Thank you for adding this.

[image]

umut04, 2022-03-19:
Hi,

How can I turn it off for IIS?

yanwucai, 2022-03-21:
@umut04 Are you asking about how to configure IIS?

umut04, 2022-03-21:
Hi @yanwucai 

How can I disable Client-Initiated Renegotiation in IIS applications?

yanwucai, 2022-03-21:
I think you can ask the IIS community. This issue is about NetX Secure.

mirfaanprtm, 2023-12-15:
how you fix it? i want fix it in my server