eclipse-threadx/netxduo #51: CoAP standalone support from NetXDuo

srinivas1506, 2021-10-01:
Hi All,

Recently I have started using NetxDuo to write few IoT applications and I was successful in writing MQTT standalone application by referring the sample provided in git repo.

Now my requirement is CoAP standalone support. If I search for CoAP its point to LWM2M which is based on CoAP. 
Could you please help me that how can I proceed with to get standalone CoAP implementation? Any pointers would be helpful to me.

Thanks,
Srinivas.

yuxin-azrtos, 2021-10-01:
@srinivas1506 thanks for reaching out to the Azure RTOS team.  Standalone CoAP is not yet available.

srinivas1506, 2021-10-04:
@yuxin-azrtos,

Thanks for the information. By any chance are your team planning to integrate CoAP near time?

If I have to try CoAP on NetXDuo, how can I proceed? Please let me know if there are any pointers or inputs or git repo links to go ahead.

Thanks in advance,
Srinivas.

hwmaier, 2021-10-14:
I just came across this comment here https://github.com/azure-rtos/netxduo/issues/3#issuecomment-632793953 where CoAP was announced to be available by December 2020. Is there an update of the release date?

yuxin-azrtos, 2021-10-14:
@srinivas1506, @hwmaier we don't have plans for CoAP at this point (there were some discussions a while back but CoAP was not prioritized).  Please reach out to azure-rtos-support@microsoft.com and we can further help you with your needs.

closed 2021-10-14 by yuxin-azrtos

yuxin-azrtos, 2021-10-14:
CoAP is built into the lwM2M stack. However it is not exposed as a standalone module.

It's still listed here: https://docs.microsoft.com/en-us/azure/rtos/netx-duo/overview-netx-duo#coap<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdocs.microsoft.com%2Fen-us%2Fazure%2Frtos%2Fnetx-duo%2Foverview-netx-duo%23coap&data=04%7C01%7Cyuxzho%40microsoft.com%7C33320cc016cf475ff5d908d98eba2935%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637697752503090153%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=GUYGb03W%2B2gUl8bQYxLsVaMWnQkUoiI8ADtMr%2BM9J1Q%3D&reserved=0> as a supported protocol.

-