eclipse-threadx/netxduo #59: Disable TLS 1.2 cause build error

DoudFPV, 2021-11-19:
I have a project where I only need TLS 1.3.

To save some space I tried to disable TLS 1.2 by : 

_#define NX_SECURE_TLS_TLS_1_2_ENABLED                   (0)_

But it causes build error in tls 1.3 client handshake files : 

`netxduo/nx_secure/src/nx_secure_tls_1_3_client_handshake.c: In function '_nx_secure_tls_1_3_client_handshake':
netxduo/nx_secure/src**/nx_secure_tls_1_3_client_handshake.c:358:68:** error: 'NX_SECURE_TLS_CRYPTO' {aka 'struct NX_SECURE_TLS_CRYPTO_STRUCT'} has no member named 'nx_secure_tls_handshake_hash_sha256_method'
  358 |             method_ptr = tls_session -> nx_secure_tls_crypto_table -> nx_secure_tls_handshake_hash_sha256_method;
      |   `

Someone know a way to solve this issue ?

tstapko, 2021-11-19:
Hello! It appears that this indeed an issue. I'm going to create an internal bug report to track it and hopefully have it resolved in the next release. 

In the meantime, you can try looking in nx_secure_tls.h, around line 1010 where the nx_secure_tls_handshake_hash_sha256_method field is defined. Remove the #if (NX_SECURE_TLS_TLS_1_2_ENABLED) and corresponding #endif. If you are using the default ciphersuite table (nx_crypto_generic_ciphersuites.c) you'll also need to remove the macro conditional in the cipersuite table initialization, around line 270. There's no guarantee this will work, but if it does you should be able to continue with TLS 1.3 only.

DoudFPV, 2021-11-19:
Thanks for you quick reply.

closed 2021-11-19 by DoudFPV