eclipse-threadx/netxduo #60: DTLS memory leak of 1 packet on each connection

DoudFPV, 2021-12-06:
**Setup description :** 
We are testing the netxduo (6.1.9) with a DTLS server.
The DTLS server code is based on the documentation ( https://docs.microsoft.com/fr-fr/azure/rtos/netx-duo/netx-secure-dtls/chapter2#small-example-system-dtls-server ) with the updates below : 
 - The TLS cipher is EC, so nx_secure_dtls_ecc_initialize() is called before nx_secure_dtls_server_session_start().
 - dtls_server_connect_notify() and dtls_server_receive_notify() use a queue to send messages to DTLS server task.

On client side, we use openssl : "openssl s_client -connect 192.168.2.2:4321 -dtls -CAfile ca.crt"

**Issue** 
After each connection/disconnection, we can see a memory leak of 1 packet (nx_packet_pool_available decreases by one) after each connection.
According to my tests, the memory leak seems to appear during the DTLS handshake.

Coincidentally, I found that adding a delay in nx_secure_tls_packet_release() before calling nx_packet_release() solves the memory leak.

tstapko, 2021-12-06:
Thanks for reporting this issue. I've created an internal tracking work item - we'll try to replicate your problem and let you know if we have any questions.

DoudFPV, 2021-12-09:
Thanks @tstapko.

Yes, there is also a memory leak here, but I already found and fix it before post this issue. 

I think there is an other memory leak during handshake, because the memory appear even without receiving or sending any datas.

Below the pool stats : 

> **Before a DTLS client connect :** 
>     IP Free TCP    : 15 min - 18 now/ 22
> 
> **When client is connected (but doesn't send any datas)**
>     IP Free TCP    : 13 min - 17 now/ 22
> 
> **After client disconnect**
>     IP Free TCP    : 13 min - 17 now/ 22

_Note : If I connect again, the value "now" decrease by now after each connexion._

If I add an osDelay(10); in nx_secure_tls_packet_release() before it calls nx_packet_release(), the memory leak disappear :

```
UINT nx_secure_dtls_packet_release_delayed(NX_PACKET *packet_ptr);
#define nx_secure_tls_packet_release nx_secure_dtls_packet_release_delayed
```
```

/**
 * ******************************************************************************
 *  Function Name: nx_secure_dtls_packet_release_delayed()
 * *******************************************************************************
 */
UINT nx_secure_dtls_packet_release_delayed(NX_PACKET *packet_ptr)
{
    osDelay(10);
    return nx_packet_release(packet_ptr);
}

```

_After adding delay, the pool stats becomes : (memory leak is fixed)_ 

> **Before a DTLS client connect :** 
>     IP Free TCP    : 16 min - **18 now**/ 22
> 
> **When client is connected (but doesn't send any datas)**
>     IP Free TCP    : 13 min - **18 now**/ 22
> 
> **After client disconnect**
>     IP Free TCP    : 13 min - **18 now**/ 22

DoudFPV, 2021-12-13:
Yes.
It is on an STM32F4, the driver is from ST : https://github.com/STMicroelectronics/x-cube-azrtos-f4

For the breakpoint : Yes, when I open the connection the break point fire.

DoudFPV, 2021-12-13:
Just a note : nx_secure_dtls_server_session_start() is called in dtls_server_thread and not in IP/driver thread, can it cause issue ?

DoudFPV, 2021-12-14:
@yanwucai Instead of adding a delay to each call to nx_secure_tls_packet_release(), I try to find the calls where they are needed.
I found that adding a delay [here](https://github.com/azure-rtos/netxduo/blob/master/nx_secure/src/nx_secure_dtls_retransmit_queue_flush.c#L119) is all that is needed to solve the memory leak : 
[image]

**As your first hunch, it seems that the problem is related to the ethernet driver.** 
_Note : on the STM32F4 driver, DMA is used to send packets, and an IT is generated when the packet is sent._

Any ideas for this problem properly?

yanwucai, 2021-12-14:
It won't cause issue if nx_secure_dtls_server_session_start() is called in dtls_server_thread.

Does the IP thread have higher priority than the DTLS server thread?

DoudFPV, 2021-12-14:
Adding the delay [here](https://github.com/azure-rtos/netxduo/blob/185bca73cc6cafbbc612ed94d5866502a020f29c/nx_secure/src/nx_secure_dtls_session_start.c#L252) at the end of handshake solve also the leak issue.
[image]

It seems that calling _nx_secure_dtls_retransmit_queue_flush() before all packet were sent (or while a packet is sent) cause the memory leak.

DoudFPV, 2021-12-14:
@yanwucai, I think I found the issue, but I don't know how to fix it properly :

On STM32F4 driver :
 - DMA is used to transfer packet. (So IP/Driver thread doesn't wait the end of transfert)
 - An IT fire at the end of DMA transfert and set the flag NX_DRIVER_DEFERRED_PACKET_TRANSMITTED.
 It causes IP/Driver execution, which call _nx_packet_transmit_release() where nx_packet_queue_next is set to NX_DRIVER_TX_DONE as below :
 
[image]

- When nx_packet_queue_next is set to NX_DRIVER_TX_DONE, calling _nx_secure_dtls_retransmit_queue_flush() will release the buffer.
[image]

But in my case, _nx_secure_dtls_retransmit_queue_flush() is certainly called BEFORE the end of transfert, so nx_packet_queue_next is NOT set to NX_DRIVER_TX_DONE yet and packet isn't released....

_It also explain why the delay solve this issue._

Is their a way to wait all packets be sent before calling _nx_secure_dtls_retransmit_queue_flush() at the end of the DTLS handshake ? (Instead of the osDelay(10))

yanwucai, 2021-12-20:
Can you replace [Line 94 - 96 ](https://github.com/azure-rtos/netxduo/blob/185bca73cc6cafbbc612ed94d5866502a020f29c/nx_secure/src/nx_secure_dtls_retransmit_queue_flush.c#L94-L96)in nx_secure_dtls_retransmit_queue_flush.c with the following lines and see if it can solve the issue?

```
    while (packet_ptr &&
           (packet_ptr != (NX_PACKET *)NX_PACKET_ENQUEUED))
```

DoudFPV, 2022-01-19:
Thanks for this reply, I will test on the board beginning of next week

DoudFPV, 2022-01-27:
@yanwucai : your fix works perfectly, thanks.

closed 2022-01-27 by DoudFPV