eclipse-threadx/netxduo #63: The block mode implementation for FTP looks like supports only full file size.

RishaHolla, 2021-12-13:
If we select FTP write mode as block mode, we need to call api to set the file size. There is a variable used 
ftp_client_ptr -> nx_ftp_client_block_remaining_size. This variable is set to
ftp_client_ptr -> nx_ftp_client_block_remaining_size = file_size; in _nx_ftp_client_file_size_set() api. 

In _nx_ftp_client_file_write() api we substract the file_size being written from ftp_client_ptr -> nx_ftp_client_block_remaining_size and when we reach the value 0, we send EOF in block header to server.

if (ftp_client_ptr -> nx_ftp_client_transfer_mode == NX_FTP_TRANSFER_MODE_BLOCK)
    {

        **/* Update the file size.  */
        ftp_client_ptr -> nx_ftp_client_block_remaining_size -= file_size;

        /* Check if the file is already sent.  */
        if (ftp_client_ptr -> nx_ftp_client_block_remaining_size == 0)**
        {

            /* Send end block header for file size.  */
            status = _nx_ftp_client_block_header_send(ftp_client_ptr, 0);
        }

This works fine, if we do a call to _nx_ftp_client_file_size_set() only once in the beginning of the file write operation with the total file size being written. Then, every time a block is written block_size( file_size in this context of _nx_ftp_client_file_write()) keeps getting substracted from nx_ftp_client_block_remaining_size and when file ends EOF is written to server. However, as we need to send block_header preceeding every block we will need to call _nx_ftp_client_file_size_set() for every block, then every time before a block is written in _nx_ftp_client_file_size_set() api ftp_client_ptr -> nx_ftp_client_block_remaining_size value is overwritten with the current block_size and not the full file_size and hence for every block the above highlighted condition is met and with every block EOF is sent. Does this mean current implementation supports only whole file as one block? splitting of file into smaller multiple block size is not supported yet?

wenhui-xie, 2021-12-21:
Yes, only support the total file size transferred in one block currently, but one file can be split into several small packets by calling _nx_ftp_client_file_write() multiple times. 
What's the scenario that you need to send multiple blocks?

bo-ms, 2022-01-20:
Closing, feel free to re-open if you have further issues.

closed 2022-01-20 by bo-ms