eclipse-threadx/netxduo #64: Call to nx_ip_create failing because of NX_IP size mismatch

josesimoes, 2021-12-14:
Hi,

My code is failing in the call to `nx_ip_create()` and I've traced the issue to a size mismatch with `ip_control_block_size` and `sizeof(NX_IP)`.
(this is check is at the very beginning of `nx_ip_create()` )

I've triple check the samples and I can't spot anything obvious...
Guessing this is probably because of the nx_user options but I fail to pinpoint the root cause and I'm out ideas on how to find that.
Any hints, please, on what I should do to debug this?

jdeere5220, 2021-12-14:
@josesimoes as you stated their are several members of the NX_IP  structure that are included or not included based on #defines, and these #defines normally come from either the nx_user.h header file or possibly the compiler command line options. Are you certain the same nx_user.h header and command line are being used when you build the netxduo library as are being used when you build your application? That's really the only way I'm aware of that this can happen, i.e. using different headers when building the library vs. building your application. As far as debugging the problem I don't have any secrets for you but maybe someone else on our team can chime in.

josesimoes, 2021-12-14:
Your comment above prompted my to dig into the build options. I ended up having to add `-DNX_INCLUDE_USER_DEFINE_FILE` to the other library that includes "nx_api.h" (and friends) and now `NX_IP` size validation in `nx_ip_create()` passes.

This maybe pointing out a possible flaw in the mechanism that includes nx_user.h in the build... Seems that the includes in the other library isn't picking up the nx_user.h and therefore `NX_IP` "looks" different for those libraries.

closed 2021-12-14 by TiejunMS