eclipse-threadx/netxduo #65: Define NX_BSD_ENABLE_DNS in build system

josesimoes, 2021-12-14:
Unless I'm missing something the only way to set this define is by editing nxd_bsd.h
This is not practical, especially on a CI-CD pipeline.
There should be an easy way of being able to set this define through the build system. Either by setting a CMake variable, or being able to grab it from nx_user.h (which I've tried and failed because I ended up getting a redefined compiler error), or... 😉

TiejunMS, 2021-12-17:
@josesimoes , did you try to add `NX_BSD_ENABLE_DNS` to `nx_user.h`? And to include `nx_user.h`, `NX_INCLUDE_USER_DEFINE_FILE` needs to be defined in your project. 

The include path should be, `nxd_bsd.c` -> `nx_api.h` -> `nx_port.h` -> `nx_user.h`.

josesimoes, 2021-12-18:
I've placed `NX_BSD_ENABLE_DNS` in `nx_user.h` and now it's working _BUT_ I had to add -DNX_INCLUDE_USER_DEFINE_FILE in the build for other library that was referencing it, as you've suggested.

closed 2021-12-18 by josesimoes