eclipse-threadx/netxduo #66: Using signed arithmetic leads to wrong results

Jerryxia32, 2021-12-16:
Hi. I found that in `nx_tcp_socket_packet_process.c` the code uses signed arithmetic to implement the logic for Page 69, RFC 793 of TCP:
```c
            else if (((INT)packet_sequence - (INT)rx_sequence >= 0) &&
                     ((INT)rx_sequence + (INT)rx_window - (INT)packet_sequence > 0))
            {
                outside_of_window = NX_FALSE;
            }
```
This does not look right to me, since the RFC states that sequence numbers are within 0 - 2\*\*32-1 and are always modulo 2\*\*32, which means they should be kept as unsigned during comparison. What's worse is that C signed overflow is undefined so the compiler is allowed to freely reorder the comparisons here.

This produces wrong results for the clang RISC-V compiler. With
```plain
packet_sequence = 0x7ffffffc, rx_sequence = 0x7ffffffc, rx_window = 0x40
```
The TCP spec on page 69 of RFC 793 says the `else if` should succeed. However, the current netx code fails on RISC-V with these inputs, causing it to fall outside of the window. Changing it into
```c
            else if ((packet_sequence >= rx_sequence) &&
                     (rx_sequence + rx_window > packet_sequence))
            {
                outside_of_window = NX_FALSE;
            }
```
fixes the issue.

May I know why the current netx code does signed arithmetic and compares to 0? Thank you.

TiejunMS, 2021-12-17:
@Jerryxia32 , thanks for sharing the details. Here is the original test from Page 69, RFC 793.
```
RCV.NXT =< SEG.SEQ < RCV.NXT+RCV.WND
```
This test makes sure the beginning sequence of TCP packet is inside the receive window. As you know, TCP sequence wraps on 2\*\*32. This *test* is not programming statement. While in NetX, we will take 2\*\*32 wrap into consideration. Here is an example of **inside** the window.
```
packet_sequence = 1
rx_sequence = 0xFFFFFFFF
rx_window = 1024
```
Using signed integer is helpful to simplify the logic of doing comparison without checking wrap. This sample data will fail the check you proposed.

We are not aware of the different behaviors of compilers. We will look into this issue, especially for RISC-V as you mentioned.

TiejunMS, 2021-12-17:
@Jerryxia32 , I wrote a piece of code for testing. Your example data are used for test.
```
int func() {
    unsigned int packet_sequence = 0x7ffffffc;
    unsigned int rx_sequence = 0x7ffffffc;
    unsigned int rx_window = 0x40;
    unsigned int outside_of_window = 1;

    if (((int)packet_sequence - (int)rx_sequence >= 0) &&
        ((int)rx_sequence + (int)rx_window - (int)packet_sequence > 0))
    {
        outside_of_window = 0;
    }

    return outside_of_window;
}
```
When the above code is compiled with RISC-V rv32gc clang 13.0.0 with -O3, the assemble code is,
```
func:                                   # @func
        mv      a0, zero
        ret
```
You can find the result here: https://godbolt.org/z/M1ze6a7h9
It does look like the logic from `nx_tcp_socket_packet_process.c` is correct. I also tried several other compilers and the result is always correct.

Could you share your compiler version, IDE and the board you are using?

Jerryxia32, 2021-12-17:
Thanks for looking into this. I also cannot reproduce this with your simple example with my compiler, but this is the code generated for `nx_tcp_socket_packet_process()`. My compiler is the stock clang compiler `Debian clang version 11.0.1-2` from Debian Bullseye. I'm not using any IDE.
```plain
;         if (packet_data_length == 0)
      4a: 8d ca         beqz    a3, 0x7c <_nx_tcp_socket_packet_process+0x7c>
;             if ((rx_window > 0) &&
      4c: b9 e3         bnez    a5, 0x92 <_nx_tcp_socket_packet_process+0x92>
;                 ((((INT)packet_sequence - (INT)rx_sequence >= 0) &&
      4e: b3 27 c8 00   slt     a5, a6, a2
      52: 93 c7 17 00   xori    a5, a5, 1
      56: 32 97         add     a4, a4, a2
      58: b3 25 e8 00   slt     a1, a6, a4
```
Above is the generated machine code. The biggest problem is that signed overflow in C is undefined so the compiler assumes overflow does not happen and freely reorders the comparisons. The instructions at `0x4e` and `0x58` are reordered into
```c
            else if (((INT)packet_sequence >= (INT)rx_sequence) &&
                     ((INT)rx_sequence + (INT)rx_window > (INT)packet_sequence))
            {
                outside_of_window = NX_FALSE;
            }
```
Then, since they are cast to `INT`, the signed version `slt` is used, so anything above `0x7fffffff` is negative and will be smaller than the other side. This is why any packet larger than `0x7fffffff` will fail. I can see that the `INT` cast is trying to capture the wrap around case, but C signed integer overflow or wrap around is undefined and it's probably not a good idea to use undefined behaviour to do that.

The way I'm reading the RFC is that your case
```plain
packet_sequence = 1
rx_sequence = 0xFFFFFFFF
rx_window = 1024
```
is supposed to fail, and we have to handle it somehow, so my code still works. Although I'm not a TCP expert and I don't know if my interpretation is correct.

If we really want to capture wrap around, how about avoiding signed arithmetic:
```c
            else if (((INT)(packet_sequence - rx_sequence) >= 0) &&
                     ((INT)(rx_sequence + rx_window - packet_sequence) > 0))
            {
                outside_of_window = NX_FALSE;
            }
```
This compiles to:
```plain
;             else if (((INT)(packet_sequence - rx_sequence) >= 0) &&
      80: b3 06 c8 40   sub     a3, a6, a2
      84: 63 c8 06 00   bltz    a3, 0x94 <_nx_tcp_socket_packet_process+0x94>
;                      ((INT)(rx_sequence + rx_window - packet_sequence) > 0))
      88: 3a 96         add     a2, a2, a4
      8a: 33 06 06 41   sub     a2, a2, a6
      8e: b3 26 c0 00   sgtz    a3, a2
```
No reordering is happening, and signed operation is only limited to the comparison.

TiejunMS, 2021-12-20:
Thanks @Jerryxia32 ! We will fix this issue in next patch release as you suggested.

For the following case,
```
packet_sequence = 1
rx_sequence = 0xFFFFFFFF
rx_window = 1024
```
the packet sequence is inside the receive window. All acceptable sequences are 0xFFFFFFFF and 0~1022.

TiejunMS, 2021-12-21:
Fixed internally. If you want to try it out before next release, please send an email to azure-rtos-support@microsoft.com.

closed 2021-12-21 by TiejunMS