eclipse-threadx/netxduo #68: Azure IoT with DPS

hmckinlay, 2021-12-16:
Has anyone got the azure-iot addon working using DPS to connect to Azure Central? 

Running a STM32 NUCLEO-H753ZI here with the latest libs [azure-rtos/netxduo](https://github.com/azure-rtos/netxduo)/[azure-sdk](https://github.com/Azure/azure-sdk-for-c/tree/main/sdk/docs/iot). Am able to bypass DPS and connect fine using SAS tokens. Once connected telemetry and direct methods are working great.

tstapko, 2021-12-16:
Not sure what tools you are using, but the "samples" section of the Azure RTOS repo (https://github.com/azure-rtos/samples) has some more detailed examples regarding DPS connectivity. Note that DPS has to be configured to recognize and assign your device to your hub instance, and the device needs to be configured with the DPS endpoint. 

It's not for the exact device you're using, but check out the example application "/stm32f746g-disco/iar/sample_azure_iot_embedded_sdk" in https://github.com/azure-rtos/samples/releases/download/v6.1_rel/Azure_RTOS_6.1_STM32L4+-DISCO_IAR_Samples_2021_11_03.zip. The DPS code isn't specific to the IDE so you should be able to see what is done to connect to a Hub instance via DPS.

hmckinlay, 2021-12-16:
I am using STM32CubeIDE and am familiar with those examples. 

Can you please confirm this addon actually works with DPS though? 

I would be happy to publish and share my source if that helps.

tstapko, 2021-12-17:
I can confirm that I've connected through DPS multiple times with the code in that sample on a couple different devices. 

I would be curious to know some details about your exact DPS failure - in my experience most DPS issues are configuration problems: device isn't properly registered, device certificate is incompatible with DPS (DPS is very picky about the X.509 Common Name/DeviceID), the device isn't assigned to a Hub instance, etc. 

Note that you can also use DPS without X.509 if you're having problems with certs - when you add an individual enrollment you can select "Symmetric Key".

TiejunMS, 2021-12-17:
@hmckinlay , as you are using Azure RTOS to connect to DPS, we have provided a convenience layer for Azure IoT SDK. You can found the specific doc [here](https://github.com/azure-rtos/netxduo/blob/master/addons/azure_iot/samples/README.md#use-device-provisioning-service-with-symmetric-key). And in each [sample ](https://github.com/azure-rtos/samples) project as @tstapko mentioned, there are also user guides in docs folder from zip file. Please let us know if it helps.

hmckinlay, 2021-12-17:
Thanks. I am using the Azure IoT SDK. 

It works fine if i bypass DPS, I do this by running the script on [this forum](https://stackoverflow.com/questions/63033615/connect-real-device-to-azure-iot-central-using-mqtt).

Below is the debugger output and error codes I am getting:
[image]

The sym key comes from 'az iot central device compute-device-key' command as per [this IoT Central doc ](https://docs.microsoft.com/en-us/azure/iot-central/core/concepts-get-connected)

`static UINT Aziot_Dps_Entry(UCHAR **iothub_hostname, UINT *iothub_hostname_length, UCHAR **iothub_device_id, UINT *iothub_device_id_length)
{
	UINT status;

	printf("I:Start Provisioning Client...\r\n");
	printf("I:Device Symmetric Key = %s, SIZE = %u\r\n", NX_AZIOT_DEVICE_ID_2_SYMMETRIC_KEY, sizeof(NX_AZIOT_DEVICE_ID_2_SYMMETRIC_KEY) - 1);
	printf("I:Endpoint = %s, SIZE = %u\r\n", NX_AZIOT_ENDPOINT, sizeof(NX_AZIOT_ENDPOINT) - 1);
	printf("I:ID Scope = %s, SIZE = %u\r\n", NX_AZIOT_ID_SCOPE, sizeof(NX_AZIOT_ID_SCOPE) - 1);
	printf("I:Registration ID = %s, SIZE = %u\r\n", NX_AZIOT_REGISTRATION_ID, sizeof(NX_AZIOT_REGISTRATION_ID) - 1);
	printf("I:PNP DPS Payload = %s, SIZE = %u\r\n", NX_AZIOT_PNP_DPS_PAYLOAD, sizeof(NX_AZIOT_PNP_DPS_PAYLOAD) - 1);

	// Initialize IoT provisioning client.
	if ((status = nx_azure_iot_provisioning_client_initialize(&nx.azu.prov_client, &nx.azu.azure,(UCHAR *)NX_AZIOT_ENDPOINT, sizeof(NX_AZIOT_ENDPOINT) - 1,(UCHAR *)NX_AZIOT_ID_SCOPE, sizeof(NX_AZIOT_ID_SCOPE) - 1,(UCHAR *)NX_AZIOT_REGISTRATION_ID, sizeof(NX_AZIOT_REGISTRATION_ID) - 1,_nx_azure_iot_tls_supported_crypto,_nx_azure_iot_tls_supported_crypto_size,_nx_azure_iot_tls_ciphersuite_map,_nx_azure_iot_tls_ciphersuite_map_size,nx.azu.tls_metadata_buffer,sizeof(nx.azu.tls_metadata_buffer),&nx.azu.root_ca_cert)))
	{
		printf("Failed on nx_azure_iot_provisioning_client_initialize!: error code = 0x%08x\r\n", status);
		return(status);
	}

	// Initialize length of hostname and device ID.
	*iothub_hostname_length = NX_AZIOT_CHAR_BUF_SZ;
	*iothub_device_id_length = NX_AZIOT_CHAR_BUF_SZ;

	// Add more CA certificates.
	if ((status = nx_azure_iot_provisioning_client_trusted_cert_add(&nx.azu.prov_client, &nx.azu.root_ca_cert_2)))
	{
		printf("Failed on nx_azure_iot_provisioning_client_trusted_cert_add!: error code = 0x%08x\r\n", status);
	}
	else if ((status = nx_azure_iot_provisioning_client_trusted_cert_add(&nx.azu.prov_client, &nx.azu.root_ca_cert_3)))
	{
		printf("Failed on nx_azure_iot_provisioning_client_trusted_cert_add!: error code = 0x%08x\r\n", status);
	}

	if(status == NX_AZURE_IOT_SUCCESS)
	{
		if(nx.azu.deviceCert)
		{
			// Initialize the device certificate.
			if ((status = nx_secure_x509_certificate_initialize(&nx.azu.device_cert, (UCHAR *)aziot_device_cert_ptr, (USHORT)aziot_device_cert_len, NX_NULL, 0,(UCHAR *)aziot_device_private_key_ptr, (USHORT)aziot_device_private_key_len , nx.azu.device_key_type)))
			{
				printf("Failed on nx_secure_x509_certificate_initialize!: error code = 0x%08x\r\n", status);
			}
			else if ((status = nx_azure_iot_provisioning_client_device_cert_set(&nx.azu.prov_client, &nx.azu.device_cert))) // Set device certificate.
			{
				printf("Failed on nx_azure_iot_provisioning_client_device_cert_set!: error code = 0x%08x\r\n", status);
			}
		}
		else
		{
			// Set symmetric key.
			if ((status = nx_azure_iot_provisioning_client_symmetric_key_set(&nx.azu.prov_client, (UCHAR *)NX_AZIOT_DEVICE_ID_2_SYMMETRIC_KEY, sizeof(NX_AZIOT_DEVICE_ID_2_SYMMETRIC_KEY) - 1)))
			{
				printf("Failed on nx_azure_iot_hub_client_symmetric_key_set!: error code = 0x%08x\r\n", status);
			}
		}
	}

	if(status == NX_AZURE_IOT_SUCCESS)
	{
		if ((status = nx_azure_iot_provisioning_client_registration_payload_set(&nx.azu.prov_client, (UCHAR *)NX_AZIOT_PNP_DPS_PAYLOAD, sizeof(NX_AZIOT_PNP_DPS_PAYLOAD) - 1)))
		{
		  printf("Failed on nx_azure_iot_provisioning_client_registration_payload_set!: error code = 0x%08x\r\n", status);
		  Error_Handler();
		}
		// Register device
		else if ((status = nx_azure_iot_provisioning_client_register(&nx.azu.prov_client, NX_WAIT_FOREVER)))
		{
			printf("Failed on nx_azure_iot_provisioning_client_register!: error code = 0x%08x\r\n", status);
			Error_Handler();
		}
		// Get Device info
		else if ((status = nx_azure_iot_provisioning_client_iothub_device_info_get(&nx.azu.prov_client,nx.azu.dps_iothub_hostname, iothub_hostname_length,nx.azu.dps_iothub_device_id, iothub_device_id_length)))
		{
			printf("Failed on nx_azure_iot_provisioning_client_iothub_device_info_get!: error code = 0x%08x\r\n", status);
			Error_Handler();
		}
		else
		{
			*iothub_hostname = nx.azu.dps_iothub_hostname;
			*iothub_device_id = nx.azu.dps_iothub_device_id;
			printf("Registered Device Successfully.\r\n");
		}
	}

	nx_azure_iot_provisioning_client_deinitialize(&nx.azu.prov_client); // Destroy Provisioning Client.

	return(status);
}`

TiejunMS, 2021-12-17:
What kind of enrollment did you use? Individual or group enrollment? For beginners, I would suggest individual enrollment as you can copy SAS from IoT central directly. The computation of key is needed for group enrollment only. If you are using group enrollment, could you try individual enrollment first to validate the IoT central connection?

hmckinlay, 2021-12-17:
I was using group enrollment. I swapped to individual and get the same error.

[image]

TiejunMS, 2021-12-17:
Did you copy the SAS token from `az` command line directly for individual enrollment?

hmckinlay, 2021-12-19:
> Did you copy the SAS token from `az` command line directly for individual enrollment?

The key was generated from powershell via the command on the next line as per the [instructions here](https://docs.microsoft.com/en-us/azure/iot-central/core/concepts-get-connected)
`openssl rand -base64 64`

The open SSL version is OpenSSL 1.1.1g  21 Apr 2020
[image]

Then copied into the from there to notepad, cleaned the space from the new line then copied into device connect page: 
[image]

When the shared key was used there is a copy button in the azure central interface that was used.

TiejunMS, 2021-12-20:
Could you send your credentials for **TEST PURPOSE** to azure-rtos-support@microsoft.com? We would like to validate if the issue is bound to your IoT Central or application.

Following data are needed:
- ID scope
- Device ID
- Primary key

NOTE: Do not reply your credentials here. It is publicly visible.

TiejunMS, 2022-01-06:
Closed as no response. Feel free to reopen it.

hmckinlay, 2022-01-21:
Hi Tiejun, 
I just sent the details through, Excuse the delay just got back from break. I don't seem to be able to re-open?
Cheers
Hamish.

TiejunMS, 2022-01-21:
Hi @hmckinlay , I'm able to connect to IoT Central using the credential you provided. Could you run the official release demo first and then port to your own application? It is possible there are misconfiguration in your stack, heap, string size or etc.

[image]

hmckinlay, 2022-01-26:
> Hi @hmckinlay , I'm able to connect to IoT Central using the credential you provided. Could you run the official release demo first and then port to your own application? It is possible there are misconfiguration in your stack, heap, string size or etc.
> 
> [image]

This is the official release demo with a few printf's added for debugging and some ifdefs removed or changed to if's.

TiejunMS, 2022-01-27:
Since I have verified the official release demo is working with the credentials you provided, could you try it again to make sure if you can connect to DPS successfully by official demo without modification? Once that is verified, you can apply your own changes one by one and see check change will fail.

closed 2022-02-07 by TiejunMS