eclipse-threadx/netxduo #69: Connecting to Azure IOT Edge device

Alovantous, 2021-12-23:
We are working to be a display based on Azure RTOS that communicates with Azure IOT Edge device. We have done sample code and built our a testing device of what we want to do with Node.js but can't figure out how to connect Azure RTOS to a Azure IOT Edge device instead of direct to cloud. Is there documentation for this or is it even support? 

In node.js we set the connection string to have the extra gatewayHostname parameter like this: 

HostName=<iothub_host_name>;DeviceId=<device_id>;SharedAccessKey=<device_key>;GatewayHostName=<edge device hostname>

which we set to the name of our IOTEdge device. But we can't seem to find how to set that.

bo-ms, 2021-12-24:
@Alovantous thanks for bringing it up, we will take a look at it and get back to you. Thanks.

wenhui-xie, 2022-01-12:
@Alovantous IoT Edge device can act as a transparent gateway referring to the guide: https://docs.microsoft.com/en-us/azure/iot-edge/how-to-create-transparent-gateway?view=iotedge-2018-06. 

And the connection string can be simplified like 
`HostName=<edge device hostname>;DeviceId=<device_id>;SharedAccessKey=<device_key>`. 
(The edge device hostname is the hostname set in the config.yaml of the Edge device.)

And for the downstream device to connect to the edge device, the edge device hostname needs to be resolvable to an IP Address, either using DNS or a host file entry on the downstream device.

bo-ms, 2022-01-20:
Closing, feel free to re-open.

closed 2022-01-20 by bo-ms