eclipse-threadx/netxduo #72: SNTP client doesn't handle KOD codes correctly

jkuwaha, 2022-01-12:
I found that SNTP client doesn't handle KOD codes correctly when NX_LITTLE_ENDIAN is defined.
Specifically, 'kiss_of_death_handler' is almost always called with NX_SNTP_KISS_OF_DEATH_PACKET instead of the correct codes like NX_SNTP_KOD_DENY, NX_SNTP_KOD_RATE, etc.

This is probably due to the NX_CHANGE_ULONG_ENDIAN macro being applied to the reference_clock_id and the Kiss code being reversed.

https://github.com/azure-rtos/netxduo/blob/185bca73cc6cafbbc612ed94d5866502a020f29c/addons/sntp/nxd_sntp_client.c#L2804-L2807

jkuwaha, 2022-01-19:
Hi @wenhui-xie
Is there any update on this issue?
If you need more information, please let me know.

wenhui-xie, 2022-01-19:
Hi @jkuwaha, I'll take a look at it and get back to you. It will be helpful if you can share me the packet trace. Thanks.

jkuwaha, 2022-01-19:
Hi @wenhui-xie, here's a screenshot. Please note that since it's difficult to let the NTP server send KOD packets, I used a script sending crafted packets.

[image]

Packet no.2 is the KOD packet I made. As the reference ID is 'DENY' (44-45-4e-59), so it is expected the kiss_of_death_handler is called with NX_SNTP_KOD_DENY, but it is actually called with NX_SNTP_KISS_OF_DEATH_PACKET.

I also found that NetX held the reference ID as 'YNED' (59-4e-45-44). So, it can be said either my crafted KOD packet or NetX's variable is wrong.  However, I think my KOD packet can be ok based on following reasons.

According to https://www.rfc-editor.org/rfc/rfc5905.html#section-7.3, 'reference ID' is interpreted depending on 'stratum';

- as a kiss code if stratum is 0
- as a reference identifier code, which is a four-octet string, if stratum is 1
- as an IPv4 address if stratum is other than above

According to https://www.rfc-editor.org/rfc/rfc5905.html#section-7.4, kiss codes are encoded in four-character ASCII strings.

And also, I found a sample capture containing a reference identifier code at https://wiki.wireshark.org/SampleCaptures.
In 'NTP_sync.pcap', there is a stratum1 packet having 'GPS' (47-50-53-00) as reference ID.  Both kiss codes and reference id codes are string, kiss codes should be stored in the same manner, I think.

wenhui-xie, 2022-01-21:
@jkuwaha Hi, thanks for your detailed information. The fix will be in the next release.

wenhui-xie, 2022-02-07:
Close as fixed. The fix is in Azure RTOS 6.1.10.

closed 2022-02-07 by TiejunMS