eclipse-threadx/netxduo #76: Question : Is MQTT QOS=2 support planned?

DoudFPV, 2022-01-27:
Is MQTT QOS=2 support planned?
(If yes any idea of the planning ?)

TiejunMS, 2022-01-28:
@DoudFPV , we don't have plan to support QoS2 at the moment. Could you share your use case which requires QoS2? If there are more and more requests to support QoS2 in the future, we will plan to do it.

DoudFPV, 2022-01-28:
@TiejunMS not sure we really need it (QoS1 might be enough).
The most critical case will be when we subscribe to a TOPIC and make an action when the value change, we worry to execute the action twice.

TiejunMS, 2022-01-28:
For this case, a local variable to track the value change is a cheaper solution compared to QoS2, while QoS2 is a generic solution to avoid message duplication.

closed 2022-02-01 by DoudFPV