eclipse-threadx/netxduo #77: Question : How to evaluate NX_PACKET_POOL size

DoudFPV, 2022-02-01:
Is there a solution to know the number of NX_PACKET_POOL necessary for an application ?

For example if the product has on ethernet with 
 - A DHCP client
 - An mDNS server
 - Two MQTT client instance with TLS 1.2 mutual auth
 - One HTTP server with TLS 1.2 

Today, for 22 packet I use : 
    #define PAYLOAD_SIZE                 1536
    #define NX_PACKET_POOL_SIZE         (( PAYLOAD_SIZE + sizeof(NX_PACKET)) * 22)
    
But 12 might be large enough, is there a way to evaluate it by design (and not by test) ?
Thanks

goldscott, 2022-02-01:
The number of packets in the pool is dependent on the driver (e.g. the number of buffer descriptors used), the application, how much activity is happening on the network, and the speed of your processor (how fast it can process packets), so it is difficult to give much guidance here.

DoudFPV, 2022-02-02:
Ok Thanks @goldscott 

I am not really sure, but on LwIP there is pool of buffers for Rx and pool of buffer for Tx, this way we are sure to have Tx buffers even when all Rx buffers are used.
On netXduo, the Rx/Tx buffer are common, does it mean that if we have a lot of incoming activity, it's possible that we can't answer because of lack of buffer?

goldscott, 2022-02-02:
It depends on your driver, but you can have two separate pools for rx and tx.