eclipse-threadx/netxduo #78: autoip + mdns and tcp server

ucayalifish, 2022-02-07:
Hi,
I'm working on test application that uses eather dhcp of autoip with mdns-sd on STM32 NUCLEO-F429ZI boad. A modified TCP server from samples starts after IP address is obtaining link-local address and serves requests.
If DHCP is used everything works as expected, but in case of AUTO IP bus fault happens after a few reqjests.
Is it a known problem? Is it a good idea to use auto ip for tcp server?

ucayalifish, 2022-02-11:
Some additional information. After playing with code bus fault disappeared. Instead, scheduler waits indefinitely on __tx_ts_wait.

TiejunMS, 2022-02-21:
@ucayalifish , are you able to recover the call stack when bus fault occurs? When scheduler waits on __tx_ts_wait, are you able to ping the IP address of device? It sounds like the issue is related to misconfiguration in application. You can try to increase thread stack, packet pool, payload size of packet. And if you are using first unused memory, try to use static memory instead.

ucayalifish, 2022-02-21:
@TiejunMS, thank you for suggestions. It helps and now I better understand how to configure NetXDuo applications! The problem was in byte pool allocation that hangs in new connection handler of TCP server. Thank you again. I'm closing the issue.

ucayalifish, 2022-02-21:
Resolved.

closed 2022-02-21 by ucayalifish