eclipse-threadx/netxduo #80: MQTT publish crashes

pawelkub, 2022-02-23:
Im working on the project in which I use ThreadX and NetXduo with mqtt client.
I have one thread (M) in which I receive mqtt data and send response for them.
In other two threads (X and Y) Im sending telemetry data via mqtt.

I use the same publish mqtt API function where I use mutex to protect access to mqtt client object for sending mqtt data for whole system. 

Everything works correctly for a while (for example 50k published messages, but the exachanged messages number are not constant), but after that I get a communication error.
One of the thread, i.e. X stucks on forever-loop when publishing mqtt telemetry. 

It stucks in function _nx_ip_header_add() on 111 line:

```
/* Assert prepend pointer is no less than data start pointer.  */
/*lint -e{946} suppress pointer subtraction, since it is necessary. */
NX_ASSERT(packet_ptr -> nx_packet_prepend_ptr >= packet_ptr -> nx_packet_data_start);
```

After that the IP instance thread stuck, and I got mqtt disconnecting.

[image]

TiejunMS, 2022-02-24:
Hi @pawelkub , could you provide the version number from nx_api.h? It usually looks like 6.1.x. The call stack shows the packet struct is corrupted. Here are quick test you can do.
1. Increase stack size.
2. Upgrade AzureRTOS to latest version.

For thread suspends on _nx_ip_header_add, could you provide the data of nx_packet_prepend_ptr, nx_packet_data_start, nx_packet_append_ptr, nx_packet_data_end and nx_packet_length from packet_ptr?

pawelkub, 2022-02-24:
Hi @TiejunMS, thanks for leply,

I used versions 6.1.7 of netxduo and threadx.
Update to latest 6.1.10 version doesn't help unfortunately.

Should I increase stack of thread which call mqtt publish function (in this case pms_thread) or mqtt client thread or IP instance thread?

packet_ptr data dump:
[image]

TiejunMS, 2022-02-28:
The packet is set incorrectly somehow. To figure out what happened, are you aware if X thread stuck first or communication error occurs first?

TiejunMS, 2022-03-14:
Close as no response.

closed 2022-03-14 by TiejunMS