eclipse-threadx/netxduo #81: Variable may be used uninitialized in _nx_mdns_service_lookup()

werfen-tom, 2022-02-23:
I am seeing a compiler warning. To a casual look, it seems like a legitimate issue. 

This is also just after one of multiple calls to `tx_mutex_put()` whose return value is not checked. If ignoring the return value is intentional, I would hope for a comment explaining why the return value is irrelevant and perhaps an explicit cast to void, i.e. `(void)tx_mutex_put(...)`. 

Regardless, returning a possibly uninitialized `status` seems incorrect.

```
nxd_mdns.c:4317:16: warning: 'status' may be used uninitialized in this function [-Wmaybe-uninitialized]
 4317 |         return (status);
      |                ^
```

bo-ms, 2022-03-04:
Hi @werfen-tom thanks for reporting this issue, we will fix it in next release.

bo-ms, 2022-03-21:
closing, @werfen-tom we already fixed it internally, the bug fix will be in next release (April).

closed 2022-03-21 by bo-ms