eclipse-threadx/netxduo #82: NX_SECURE_ENABLE_AEAD_CIPHER not defined with TLS 1.3

GKNSTM, 2022-03-02:
In netxduo 6.1.9:

if we want to use TLS 1.3 and define NX_SECURE_TLS_TLS_1_3_ENABLED in nx_user.h there is a compilation error in nx_crypto_generic_ciphersuites.c:

Error[Pe029]: expected an expression C:\work\stm32u5\Firmware\Middlewares\ST\netxduo\crypto_libraries\src\nx_crypto_generic_ciphersuites.c 205 
Error[Pe1345]: an empty initializer is invalid for an array with unspecified bound C:\work\stm32u5\Firmware\Middlewares\ST\netxduo\crypto_libraries\src\nx_crypto_generic_ciphersuites.c 197 

The code is:

    #if (NX_SECURE_TLS_TLS_1_3_ENABLED)
    NX_SECURE_TLS_CIPHERSUITE_INFO _nx_crypto_ciphersuite_lookup_table_tls_1_3[] =
    {
    #ifdef NX_SECURE_ENABLE_AEAD_CIPHER
        {TLS_AES_128_GCM_SHA256,                  &crypto_method_ecdhe,      &crypto_method_ecdsa,     &crypto_method_aes_128_gcm_16,  96,      16,        &crypto_method_sha256,         32,         &crypto_method_hkdf},
        /* SHA-384 ciphersuites not yet supported... {TLS_AES_256_GCM_SHA384,                  &crypto_method_ecdhe,      &crypto_method_rsa,     &crypto_method_aes_256_gcm_16,  16,      16,        &crypto_method_sha384,         48,         &crypto_method_hkdf},*/
        {TLS_AES_128_CCM_SHA256,                  &crypto_method_ecdhe,      &crypto_method_ecdsa,     &crypto_method_aes_ccm_16,       96,      16,        &crypto_method_sha256,         32,         &crypto_method_hkdf},
        {TLS_AES_128_CCM_8_SHA256,                &crypto_method_ecdhe,      &crypto_method_ecdsa,     &crypto_method_aes_ccm_8,       96,      16,        &crypto_method_sha256,         32,         &crypto_method_hkdf},
    #endif
    };

So the error is due to NX_SECURE_ENABLE_AEAD_CIPHER not defined.

It is supposed to be defined in nx_secure_tls.h line 130:

    #if (NX_SECURE_TLS_TLS_1_3_ENABLED) || defined(NX_SECURE_ENABLE_ECJPAKE_CIPHERSUITE)
    #ifndef NX_SECURE_ENABLE_AEAD_CIPHER
    #define NX_SECURE_ENABLE_AEAD_CIPHER
    #endif /* NX_SECURE_ENABLE_AEAD_CIPHER */
    #endif

But in nx_secure_tls.h line 130 , NX_SECURE_TLS_TLS_1_3_ENABLED is not yet defined. It will be defined later in same file at line 422:

    #ifdef NX_SECURE_TLS_ENABLE_TLS_1_3
    #define NX_SECURE_TLS_TLS_1_3_ENABLED                   (1)
    #endif 

So maybe the definition of NX_SECURE_ENABLE_AEAD_CIPHER should be placed after definition of NX_SECURE_TLS_TLS_1_3_ENABLED ?

TiejunMS, 2022-03-03:
Hi @GKNSTM , the user configuration is `NX_SECURE_TLS_ENABLE_TLS_1_3` rather than `NX_SECURE_TLS_TLS_1_3_ENABLED`. Could you try to define `NX_SECURE_TLS_ENABLE_TLS_1_3` and `NX_SECURE_ENABLE_AEAD_CIPHER` in `nx_user.h` and see if this solves the compiler issue?

GKNSTM, 2022-03-03:
Sorry, I think I mixed up `NX_SECURE_TLS_ENABLE_TLS_1_3` and `NX_SECURE_TLS_TLS_1_3_ENABLED`.

I defined `NX_SECURE_TLS_ENABLE_TLS_1_3 ` in nx_user.h and had the error for `NX_SECURE_ENABLE_AEAD_CIPHER` not defined.

To rephrase correctly:

if we want to use TLS 1.3 and define `NX_SECURE_TLS_ENABLE_TLS_1_3` in nx_user.h there is a compilation error in nx_crypto_generic_ciphersuites.c.

This is because in nx_secure_tls.h line 130, `NX_SECURE_ENABLE_AEAD_CIPHER` is not defined because `NX_SECURE_TLS_TLS_1_3_ENABLED` will be defined later in the file.

I confirm that if I define `NX_SECURE_TLS_ENABLE_TLS_1_3` and `NX_SECURE_ENABLE_AEAD_CIPHER` in nx_user.h there is no compilation error. 

But `NX_SECURE_ENABLE_AEAD_CIPHER` could be defined automatically in nx_secure_tls.h just by moving its definition from line 130-137 to after definition of `NX_SECURE_TLS_TLS_1_3_ENABLED` (line 430 for example). This would be easier for the user.

TiejunMS, 2022-03-04:
I see. We will replace line 130 from
`#if (NX_SECURE_TLS_TLS_1_3_ENABLED) || defined(NX_SECURE_ENABLE_ECJPAKE_CIPHERSUITE)`
to
`#if defined(NX_SECURE_TLS_ENABLE_TLS_1_3) || defined(NX_SECURE_ENABLE_ECJPAKE_CIPHERSUITE)`
in next release. Let me know if it helps.

GKNSTM, 2022-03-04:
Yes I think this solves the issue.

closed 2022-03-04 by TiejunMS