eclipse-threadx/netxduo #83: Request : Cipher ECDHE_PSK (for DTLS)

DoudFPV, 2022-03-04:
We use a DTLS connexion with the cipher TLS_PSK_WITH_AES_128_GCM_SHA256 and it's works great, thanks to the netxduo team !

We have a request to use ECDHE_PSK,  Does it possible on netXduo? (Like MBEDTLS_TLS_ECDHE_PSK_WITH_AES_128_CBC_SHA256 )

Thanks

TiejunMS, 2022-03-07:
Hi @DoudFPV , ECDHE-PSK is not supported by NetXDuo. Could you share with us your application protocol and use case? Thanks!

DoudFPV, 2022-03-07:
PSK is used to secure the connection of IOT devices.
According to our security manager, "simple" PSK is not secure enough and we need to implement ECDHE-PSK.
It would be great if ECDHE-PSK could be implemented in netXduo, otherwise I'm afraid we'll have to revert to LwIP/mbedTLS.

TiejunMS, 2022-03-09:
@DoudFPV , I'm sorry to tell that we don't have plan to support it in the near future.

closed 2022-03-09 by DoudFPV