eclipse-threadx/netxduo #86: sample_azure_iot_embedded_sdk_with_retry.c returns MQTT CONNECT FAIL status:65541

dsilva-vd, 2022-03-23:
NETXDUO 6.1.7 - Tied to latest NETX distributed by STM32Cube IDE.

Using sample_azure_iot_embedded_sdk_with_retry.c I am consistently getting MQTT CONNECT FAIL status:65541 (0x10005) when trying to connect to my IOT Hub. The IOT hub is available as other clients (Python Azure SDK) can connect to the Hub).  Is there a way to turn on more debugging or some idea of where to start debugging this. 

Nx_SNTP_Client application started..
STM32 IpAddress: XXX.XXX.XXX.XXX 
SNTP client connected to NTP server : < ca.pool.ntp.org > 

SNTP update :
Mar 23, 2022 22:8:18.859 UTC 

23-03-2022 / 22:08:18
[INFO] Azure IoT Security Module has been enabled, status=0
IoTHub Host Name: Myazurehub.azure-devices.net; Device ID: mydevice.
[ERROR] IoTHub client connect fail: MQTT CONNECT FAIL status: 65541
Disconnected from IoTHub!: error code = 0x00010005
Failed on nx_azure_iot_hub_client_connect!
reconnecting iothub, after backoff
[ERROR] IoTHub client connect fail: MQTT CONNECT FAIL status: 65541
Disconnected from IoTHub!: error code = 0x00010005
reconnecting iothub, after backoff
[ERROR] IoTHub client connect fail: MQTT CONNECT FAIL status: 65541
Disconnected from IoTHub!: error code = 0x00010005
reconnecting iothub, after backoff

dsilva-vd, 2022-03-23:
I have also  tried the secondary SAS key and the results are the same.

dsilva-vd, 2022-03-23:
I have attached my app_netxduo.c file which sets up the IP instance and SNTP client before calling the sample_entry.

`
/* USER CODE BEGIN Header */
/**
  ******************************************************************************
  * @file    app_netxduo.c
  * @author  MCD Application Team
  * @brief   NetXDuo applicative file
  ******************************************************************************
  * @attention
  *
  * Copyright (c) 2021 STMicroelectronics.
  * All rights reserved.
  *
  * This software is licensed under terms that can be found in the LICENSE file
  * in the root directory of this software component.
  * If no LICENSE file comes with this software, it is provided AS-IS.
  *
  ******************************************************************************
  */
/* USER CODE END Header */

/* Includes ------------------------------------------------------------------*/
#include "app_netxduo.h"

/* Private includes ----------------------------------------------------------*/
/* USER CODE BEGIN Includes */
#include "nx_ip.h"
#include "stm32f4xx_hal_rtc.h"

#include "nx_secure_tls_api.h"
#include <time.h>
/* USER CODE END Includes */

/* Private typedef -----------------------------------------------------------*/
/* USER CODE BEGIN PTD */
TX_THREAD AppMainThread;
TX_THREAD AppSNTPThread;
TX_THREAD AppIoTClientThread;

TX_SEMAPHORE Semaphore;

NX_PACKET_POOL           AppPool;

NX_IP                    IpInstance;
NX_DHCP                  DhcpClient;
NX_DNS                   DnsClient;
NX_SNTP_CLIENT           SntpClient;

TX_EVENT_FLAGS_GROUP     SntpFlags;

ULONG                    IpAddress;
ULONG                    NetMask;

CHAR                     buffer[64];

UCHAR                    *pointer;

struct tm timeInfos;
/* RTC handler declaration */
RTC_HandleTypeDef RtcHandle;
/* USER CODE END PTD */

/* Private define ------------------------------------------------------------*/
/* USER CODE BEGIN PD */
static UINT kiss_of_death_handler(NX_SNTP_CLIENT *client_ptr, UINT KOD_code);
static void display_rtc_time(RTC_HandleTypeDef *hrtc);
static void rtc_time_update(NX_SNTP_CLIENT *client_ptr);
static UINT unix_time_get(ULONG *unix_time);
[... 565 lines cut ...]

  uint8_t hh = RTC_Time.Hours;
  uint8_t mm = RTC_Time.Minutes;
  uint8_t ss = RTC_Time.Seconds;
  uint8_t d = RTC_Date.Date;
  uint8_t m = RTC_Date.Month;
  uint16_t y = RTC_Date.Year;
  uint16_t yr = (uint16_t)(y+2000-1900);
  time_t currentTime = {0};
  struct tm tim = {0};
  tim.tm_year = yr;
  tim.tm_mon = m - 1;
  tim.tm_mday = d;
  tim.tm_hour = hh;
  tim.tm_min = mm;
  tim.tm_sec = ss;
  currentTime = mktime(&tim);

  /* Return number of seconds since Unix Epoch (1/1/1970 00:00:00).  */
  *unix_time =  (ULONG)currentTime;

  return(NX_SUCCESS);
}
/* USER CODE END 1 */
`

dsilva-vd, 2022-03-24:
Forgot to attach the console output 

`
Nx_SNTP_Client application started..
STM32 IpAddress: XXX.XXX.X.XXX
SNTP client connected to NTP server : < fr.pool.ntp.org > 

SNTP update :
Mar 24, 2022 15:29:10.374 UTC 

24-03-2022 / 15:29:10
24-03-2022 / 15:29:10
24-03-2022 / 15:29:10
24-03-2022 / 15:29:10
24-03-2022 / 15:29:10
24-03-2022 / 15:29:10
24-03-2022 / 15:29:10
24-03-2022 / 15:29:10
[INFO] Azure IoT Security Module has been enabled, status=0
IoTHub Host Name: myhub.azure-devices.net; Device ID: stm32F429.
24-03-2022 / 15:29:10
24-03-2022 / 15:29:10
24-03-2022 / 15:29:10
24-03-2022 / 15:29:12
24-03-2022 / 15:29:12
24-03-2022 / 15:29:12
24-03-2022 / 15:29:13
24-03-2022 / 15:29:13
24-03-2022 / 15:29:13
24-03-2022 / 15:29:14
24-03-2022 / 15:29:15
24-03-2022 / 15:29:16
24-03-2022 / 15:29:16
24-03-2022 / 15:29:16
24-03-2022 / 15:29:17
24-03-2022 / 15:29:18
24-03-2022 / 15:29:19
24-03-2022 / 15:29:19
24-03-2022 / 15:29:19
24-03-2022 / 15:29:19
24-03-2022 / 15:29:20
24-03-2022 / 15:29:21
[ERROR] IoTHub client connect fail: MQTT CONNECT FAIL status: 65541
Disconnected from IoTHub!: error code = 0x00010005
Failed on nx_azure_iot_hub_client_connect!
reconnecting iothub, after backoff
24-03-2022 / 15:29:22
24-03-2022 / 15:29:22
24-03-2022 / 15:29:22
24-03-2022 / 15:29:23
24-03-2022 / 15:29:24
24-03-2022 / 15:29:25
24-03-2022 / 15:29:25
24-03-2022 / 15:29:25
24-03-2022 / 15:29:25
24-03-2022 / 15:29:25
24-03-2022 / 15:29:26
24-03-2022 / 15:29:28
24-03-2022 / 15:29:28
24-03-2022 / 15:29:28
24-03-2022 / 15:29:28
24-03-2022 / 15:29:29
24-03-2022 / 15:29:30
24-03-2022 / 15:29:31
24-03-2022 / 15:29:31
24-03-2022 / 15:29:31
24-03-2022 / 15:29:32
24-03-2022 / 15:29:33
24-03-2022 / 15:29:33
24-03-2022 / 15:29:34
24-03-2022 / 15:29:34
24-03-2022 / 15:29:34
24-03-2022 / 15:29:35
`

TiejunMS, 2022-03-25:
Hi @dsilva-vd , I saw there are only 10 packets in the pool. What is the value of RX BD? One simple test you can try is to increase packet pool count from 10 to 30 and see if the problem is solved.

dsilva-vd, 2022-03-25:
Hi @TiejunMS I increased NX_APP_MEM_POOL_SIZE to 70kBytes and increased the packet pool count to 30. This didn't change the behavior.

Just to confirm these are the changes I made:
`
#define PAYLOAD_SIZE                 1536
#define NX_PACKET_POOL_NUM_PACKETS   30
#define NX_PACKET_POOL_SIZE          (( PAYLOAD_SIZE + sizeof(NX_PACKET)) * NX_PACKET_POOL_NUM_PACKETS)
`

PACKET_POOL_SIZE is used to allocate the packet pool

What is RX BD? Can you point me to a file or name of define I can search for?
Thanks

TiejunMS, 2022-03-28:
If you are using official driver from ST, then the value is defined [here](https://github.com/STMicroelectronics/x-cube-azrtos-f4/blob/5a854b28de9e67380f65c7e2b852e3add0fb9000/Middlewares/ST/netxduo/common/drivers/ethernet/nx_stm32_eth_driver.h#L97). Since the problem is not solved after increasing the packet pool, it is not due to lack of packet. Are you able to capture packet trace by wireshark and share with us? It is helpful to understand what happens during the connection.

dsilva-vd, 2022-03-28:
Hi I will try and capture a Wireshark but as the device is a wired device I will try and find a way to do man in the middle listening.

The defines are as follows

`
#define ETH_MAX_PACKET_SIZE       1524U    /*!< ETH_HEADER + ETH_EXTRA + ETH_VLAN_TAG + ETH_MAX_ETH_PAYLOAD + ETH_CRC */
#define ETH_RX_BUF_SIZE                ETH_MAX_PACKET_SIZE /* buffer size for receive               */
#define ETH_TX_BUF_SIZE                ETH_MAX_PACKET_SIZE /* buffer size for transmit              */
#define ETH_RXBUFNB                    4U       /* 4 Rx buffers of size ETH_RX_BUF_SIZE  */
#define ETH_TXBUFNB                    4U       /* 4 Tx buffers of size ETH_TX_BUF_SIZE  */
`

dsilva-vd, 2022-03-28:
[ST32 Ethernet - bridge mode adapter.zip](https://github.com/azure-rtos/netxduo/files/8366850/ST32.Ethernet.-.bridge.mode.adapter.zip)
Hi @TiejunMS Here is a wireshark capture. The ST32 device has address 192.168.0.126. The network connectivity is bridged through 192.168.0.143 (which was running Wireshark).  

I don't see any packets from the the ST32 device indicating it is starting the MQTT connection to the Azure IoT Hub. I await your thoughts on reviewing the wireshark capture.
Thanks

TiejunMS, 2022-03-29:
@dsilva-vd , it does not look like the bridge is working. Could you double check the bridge configurations?

Btw, if you would like to share your sample_config.h with us for checking, feel free to send it to azure-rtos-support@microsoft.com

dsilva-vd, 2022-03-29:
@TiejunMS Here are the files for my iot client. I believe I renamed the sample_config.h to azure_iot_client_config.h  
[IoTClient.zip](https://github.com/azure-rtos/netxduo/files/8367762/IoTClient.zip)

I will look into the bridging configuration again tomorrow. Thanks

TiejunMS, 2022-03-29:
@dsilva-vd , could you fill in the valid hostname, device_id and symmetric key in the config.h? Just want to double check the configurations are all correct. You can delete the device once it is verified. To avoid credential leak publicly, you can send it through support email listed above.

dsilva-vd, 2022-03-29:
@TiejunMS I sent in the config file with credentials to the support email today. I'll keep working on the wireshark capture.

TiejunMS, 2022-03-30:
@dsilva-vd , I have verified the credentials you shared are valid in our official sample. If you are using the same sample, then the issue might be related to Ethernet driver. Did you port the driver or it is provided by ST? Anyway, a packet trace will be helpful to show us what happened during the connection.

dsilva-vd, 2022-03-30:
@TiejunMS The Ethernet driver is provided by ST in an example on setting up a simple SNTP client (using Ethernet). I took that example and after successfully connecting to SNTP server and getting a time update I start the Azure IoT embedded client thread.

dsilva-vd, 2022-04-04:
@TiejunMS I'm closing this issue. STM has released an AZure RTOS version of their cellular FW package and my end goal is azure connection with cellular PPP interface. The ethernet example was for temporary development. I will re-integrate the azure iot embedded client with the cellular FW package and open a new ticket if I encounter any issues.

Thanks

TiejunMS, 2022-04-06:
Thanks for letting me know. Feel free to reopen the issue if needed.

dsilva-vd, 2022-05-05:
@TiejunMS I need to reopen this issue.  

I have been able to integrate the Azure IoT client  (sample_azure_iot_embedded_skd_with_retry, Netxduo 6.1.7 release) with the Azure RTOS STM32 Cellular package.  

The IoT client behavior is exactly the same as when I was running the client over Ethernet.  The same function call returns the error value. The last time you helped troubleshoot this issue you wanted to see a Wireshark capture. That won't be simple with the cellular modem so is there another option to troubleshoot this? 

I could purchase the same reference kit used in the Netx example code i.e. B-L475E-IOT01A2 (B-L475E-IOT01A1) and run the same example to capture Wireshark but as that example works I'm not sure what it would tell us.  

if this is helpful I've attached a screenshot of the ThreadX thread list the moment the mqtt connect call is made. Perhaps there is a thread priority issue that you can see
[image]

dsilva-vd, 2022-05-05:
Forgot to reopen issue

TiejunMS, 2022-05-06:
@dsilva-vd , since you were able to reproduce the issue on both Ethernet and Cellular interfaces, I think the issue is related to application. So could you capture the packets through Ethernet which should be much simpler?

dsilva-vd, 2022-05-09:
Hi @TiejunMS I captured a wireshark capture using an passive network tap. I'm not sure I have the correct wireshark settings but here is the captured log. The device's IP is 192.168.0.138
[Ethernet Azure IoT Hub capture.zip](https://github.com/azure-rtos/netxduo/files/8655486/Ethernet.Azure.IoT.Hub.capture.zip)

dsilva-vd, 2022-05-09:
Updated capture. The first one was not captured properly and I deleted it.  I can see from the logs there is a TLS failure (Fatal Alert)
[Ethernet Azure IoT Hub capture.zip](https://github.com/azure-rtos/netxduo/files/8655773/Ethernet.Azure.IoT.Hub.capture.zip)

dsilva-vd, 2022-05-10:
Hi @TiejunMS ,

I had removed the 1st cert as I saw the comment that MS was going to transition to Digicert and removed it. After adding the cert back in the Ethernet example worked.  I also found on the ethernet sample that I needed a larger packet pool for Net X.  

I made the same changes (adding back in root cert 1) on my cellular example and successfully connected there as well.  

Thank you for your help!

closed 2022-05-10 by dsilva-vd

TiejunMS, 2022-05-11:
Good to know it is working!

bo-ms, 2022-05-12:
Hi @dsilva-vd, the start date of the Azure IoT root certificate migration from June1st 2022, to no earlier than Feb 15th, 2023. you may check here: https://techcommunity.microsoft.com/t5/internet-of-things-blog/azure-iot-tls-critical-changes-are-almost-here-and-why-you/ba-p/2393169