eclipse-threadx/netxduo #88: NX_SECURE_TLS_PREMASTER_SIZE too small for PSK 63 bytes key

DoudFPV, 2022-03-31:
With PSK, the nx_secure_tls_pre_master_secret is too small for 63 bytes PSK key.  

Default NX_SECURE_TLS_PREMASTER_SIZE is 68 or 48bytes but should be :  
```
#define NX_SECURE_TLS_PREMASTER_SIZE                      (2 + 2 + NX_SECURE_TLS_MAX_PSK_SIZE + NX_SECURE_TLS_MAX_PSK_SIZE + 1) 
```

    /* From RFC 4279:
    The premaster secret is formed as follows: if the PSK is N octets
    long, concatenate a uint16 with the value N, N zero octets, a second
    uint16 with the value N, and the PSK itself.
    |   2   |   <N>  |   2   |   <N>   |
    |   N   |    0   |   N   |   PSK   |

yanwucai, 2022-04-01:
I think you can define NX_SECURE_TLS_PREMASTER_SIZE to a larger size in nx_secure_user.h if you need long PSK keys.

DoudFPV, 2022-04-01:
Yes I did.

But today with default value, NX_SECURE_TLS_MAX_PSK_SIZE is 64, and handshake failed if the key si bigger than 33 bytes, that is why it looks like an issue for me.

yanwucai, 2022-06-15:
We have updated the default value of NX_SECURE_TLS_PREMASTER_SIZE when PSK is enabled, and the fix will be released in the next release.

DoudFPV, 2022-06-15:
Thanks I appreciate

closed 2022-06-15 by DoudFPV