eclipse-threadx/netxduo #90: DTLS/PSK issue when NX_SECURE_KEY_CLEAR is declared

DoudFPV, 2022-04-05:
DTLS works very well by default.

I tried to define NX_SECURE_KEY_CLEAR : 
- for small DTLS sessions it works well
- but I got hardfault (from packet_release) on big session, after around 22Ko of data.

Does NX_SECURE_KEY_CLEAR is supposed to work with DTLS/PSK ?

TiejunMS, 2022-04-06:
Thanks for reporting the issue, @DoudFPV! Could you share a code snippet to reproduce the issue? We will further look into it.

fdesbiens, 2026-09-25:
I’m cleaning up older issues. `NX_SECURE_KEY_CLEAR` is intended to work with DTLS/PSK, but I can’t determine the cause of the reported hard fault during packet release after about 22 KB without a reproducer. I’m closing this issue due to inactivity. If the fault still occurs on current `dev`, I’d welcome a new issue with a minimal reproducer, MCU and network driver details, packet pool size, and a fault backtrace showing the packet fields at release.

closed 2026-09-25 by fdesbiens