eclipse-threadx/netxduo #94: Socket bind/listen to specific IP

vks3, 2022-04-22:
I'm having a device with two Ethernet ports. I've set them up under a single IP instance by adding a secondary interface. Now, using either the NetX native API or the BSD API, is it possible to do a bind/listen on specific IP address only? What I need is for a TCP server to be able to listen on a specific IP only, or a TCP client to make a connect request with source address as explicitly specified during bind or listen.
An addon question to the above is that is it allowed to have multiple IP addresses assigned to a single physical port under NetX.

TiejunMS, 2022-04-24:
It is possible to bind to specific IP address by both native and BSD APIs. For BSD APIs, you can simply use `bind()`. While in native APIs, you will need to setup `nx_tcp_socket_syn_received_notify` and do the filtering in this callback function.

To support multiple IPv4 addresses, you will need to create multiple logical interfaces (`NX_INTERFACE`) and handle them correctly in network driver. See example [here](https://github.com/azure-rtos/netxduo/blob/1eee883def078a91b8cd19159d1796e3a8862814/common/src/nx_ram_network_driver.c#L155). For IPv6 addresses, each interface can have multiple addresses.

TiejunMS, 2022-05-11:
Closing. Feel free to reopen it.

closed 2022-05-11 by TiejunMS