eclipse-threadx/netxduo #96: SPI Ethernet driver or PPP over serial driver for netxduo

nilesh-dryad, 2022-04-29:
Is there any example for SPI Ethernet driver or PPPoS ? I have seen sample in this repository but there is no proper documentation or ready to use example for that.
In the sample example, it is using static IP in PPP. How can PPP be used with static IP? May be there is a different use of that but I can't find documentation on that example.

bo-ms, 2022-05-02:
Hi @nilesh-dryad [here ](https://docs.microsoft.com/en-us/azure/rtos/netx-duo/netx-duo-ppp/chapter1)is the PPP user guide. The PPP [sample](https://github.com/azure-rtos/netxduo/blob/master/samples/demo_netx_ppp.c) demonstrates the PPP client and PPP server, I guess the static IP you saw is set in PPP server, not for PPP client. We also support PPP over Ethernet ([PPPoE](https://github.com/azure-rtos/netxduo/tree/master/addons/pppoe)).

dsilva-vd, 2022-05-05:
@nilesh-dryad if you look at ST Microelectronics' x-cube-cellular FW package you can see how a PPPoS interface has been created.  Not sure what platform you have but hope this helps

closed 2022-05-19 by bo-ms