eclipse-threadx/netxduo #98: Using Azure IoT Module clients

dsilva-vd, 2022-05-13:
Hi,

This is a question on best practices using the NX Azure IoT API. My application is a cellular BLE gateway and my azure client uses module clients to represent each BLE sensor device.  What is the recommended call pattern to connect each module client?

Do I make multiple calls to `nx_azure_iot_hub_client_initialize()` and `nx_azure_iot_hub_client_connect` for the device client and each module client?  

I am using the sample_azure_iot_embedded_sdk_with_retry example.  Does the thread stack size have to be increased per extra client that is connected? 

Thanks!

bo-ms, 2022-05-16:
Hi @dsilva-vd You are right, if you making multiple connections (device connection and module connections), you may need to make multiple calls to initialize() and connect() as below. For the stack size, I do not think you need to increase it, but you are able to run your sample for verification, if it works, no need to increase the size.
```
nx_azure_iot_hub_client_initialize(iothub_client1);
nx_azure_iot_hub_client_connect(iothub_client1);

nx_azure_iot_hub_client_initialize(iothub_client2);
nx_azure_iot_hub_client_connect(iothub_client2);
```

dsilva-vd, 2022-05-26:
Thank you @bo-ms.  I followed you advice and that worked.  I am running into a separate error which I'll create a new ticket for.

closed 2022-05-26 by dsilva-vd