eclipse-threadx/netxduo #101: Question : MQTT TLS check CN

DoudFPV, 2022-05-17:
Hi,

I am using MQTT with TLS v1.2 (netxduo is client and check the server certificate).

Does netxduo check if the "CN" in certificate match with the url ? (It seems not).
Does it possible to enable this verification ?

Thanks for all your work.

tstapko, 2022-05-17:
Check out the API  [nx_secure_x509_common_name_dns_check()](https://github.com/azure-rtos/netxduo/blob/master/nx_secure/src/nx_secure_x509_common_name_dns_check.c). It should do what you want. 

It's not done by default because the NetX TLS stack isn't always paired with DNS.

Hope that helps!

TiejunMS, 2022-05-18:
Here is a reference implementation in azure iot addons.
* In tls_setup callback function of MQTT, set the certificate verify callback function [here](https://github.com/azure-rtos/netxduo/blob/1eee883def078a91b8cd19159d1796e3a8862814/addons/azure_iot/nx_azure_iot.c#L682).
* In certificate verify function, call [nx_secure_x509_common_name_dns_check](https://github.com/azure-rtos/netxduo/blob/1eee883def078a91b8cd19159d1796e3a8862814/addons/azure_iot/nx_azure_iot.c#L562) as mentioned by @tstapko .

DoudFPV, 2022-05-18:
Perfect, thanks

closed 2022-05-18 by DoudFPV