eclipse-threadx/netxduo #103: question - thread suspension during UDP send

lomer2, 2022-05-25:
The documentation states that "There is no thread suspension on sending UDP packets because all UDP packet transmissions are processed immediately.". However the UDP send service uses the IP mutex and hence sending UDP packets from high-priority threads may be suspended due to low-priority threads using the stack (the mutex is created with TX_NO_INHERIT). is there an issue with switching the mutex to a priority inheritance mutex, or is there a way to free up the UDP transmit path from using the mutex?

TiejunMS, 2022-05-26:
@lomer2, I get the situation you described. Could you describe the problem you are facing? What is the downside for none inherit mutex?

lomer2, 2022-05-26:
thanks @TiejunMS, I have a high priority thread that runs an HTTP server, and a high priority thread sending UDP packets at a high bandwidth. occasionally there is a conflict and the high priority threads gets suspended on the mutex, which could be ok by itself - but if it also coincides with a medium priority worker thread, it will get the CPU and not allow the HTTP server to release the mutex (i.e. classic priority inversion)

TiejunMS, 2022-05-27:
@lomer2, I get your point and we are aware of priority inversion. In your case, are you seeing packet drop, slow benchmark or something else with priority inversion? I would like to learn the possible issue here. And if you get a chance, could you do a local modification to use `TX_INHERIT` in [nx_ip_create.c](https://github.com/azure-rtos/netxduo/blob/1eee883def078a91b8cd19159d1796e3a8862814/common/src/nx_ip_create.c#L209)? Let's see if the issue you encountered can be solved by this change.

lomer2, 2022-05-29:
thanks @TiejunMS, the symptom is naturally packet drops.
we tried doing the local modification to use TX_INHERIT but we saw some unexpected behaviors in the system as a result, such as deadlocks in the HTTP stack. Not quite sure how it is related at this point.

TiejunMS, 2022-05-30:
Are you able to list all callstacks when your application hit the deadlock?

lomer2, 2022-05-30:
We will not continue to explore this direction for now, we will redesign the priorities of the thread to address the issue. closing for now

closed 2022-05-30 by lomer2