eclipse-threadx/netxduo #105: nx_azure_iot_json_writer_append_property_with_double_value doesnt return meaningful failure errors

ryanwinter, 2022-05-27:
As outlined in https://github.com/azure-rtos/getting-started/discussions/337, the customer had a hard to diagnosing the returned error from his call to the json writer as it returned a generic error NX_AZURE_IOT_SDK_CORE_ERROR.

Upon investigation, it seems that this function swallows all errors from the embed c-sdk. In this case the user was unable to know that they had exceeded the memory buffer, even though this is reported back to nextduo, until he debugged into the function.

The preference would be to map the underlying c-sdk errors codes to netxduo errors codes and return these to the customers.

bo-ms, 2022-05-30:
Thanks @ryanwinter for reporting this issue, we will check it.

bo-ms, 2023-03-20:
Closing. Created internal task for investigation.

closed 2023-03-20 by bo-ms