eclipse-threadx/netxduo #106: About adjusting NX_PACKET->nx_packet_prepend_ptr

rk3399, 2022-05-28:
In the eth driver, after applying for a data packet from the packet pool, it will nx_packet_prepend_ptr += sizeof(USHORT), and then use nx_packet_prepend_ptr as the buffer first address to receive data. I want to know why +sizeof(USHORT) is needed, because this will make nx_packet_prepend_ptr not aligned to 4 bytes, and if you want to use dma to transmit data, the dma controller will automatically align this address, making the data reception error!
[image]
[image]

bo-ms, 2022-05-28:
Hi @rk3399,  Thanks for asking question. The size of Ethernet header is 14 bytes, changing the prepend_ptr two bytes is to make sure TCP/IP header 4 bytes alignment.

rk3399, 2022-05-29:
> 

Thanks for the reply. I hope to pass it to the first address of the 4-byte aligned to DMA, whether it is good suggestion?

bo-ms, 2022-05-30:
Hi @rk3399 we encountered this requirement of 4-bytes alignment for some DMA, could you check if DMA supports offset? if support, you can directly set the first address as 4-bytes alignment, and set offset as 2, otherwise, you may need to prepare the buffer to DMA to receive data, then copy it into packet.

xiaocq2001, 2022-06-01:
Currently USBX Host Controller Driver (HCD) calls ST's BSP HAL_HCD_ API to issue data transfer. Maybe you can try to disable DMA in HAL_HCD_ on initialization. This way FIFO is used and unaligned address should be OK.

rk3399, 2022-06-01:
> 

> 当前 USBX 主机驱动程序 (H) 调用 ST 的 BSP HAL_API 来发出。可能您可以试用数据_HCD_初始化传输时禁用 HAL_HCD_中的 DMA。这种方式使用 FIFO 并且未命名的地址应该没有问题。

Disabling DMA does work. But this will cause CDC-ECM to occupy almost all cpu resources (each NCK will generate an interrupt), and other tasks will not be executed

rk3399, 2022-06-01:
> Currently USBX Host Controller Driver (HCD) calls ST's BSP HAL_HCD_ API to issue data transfer. Maybe you can try to disable DMA in HAL_HCD_ on initialization. This way FIFO is used and unaligned address should be OK.

Disabling DMA does work. But this will cause CDC-ECM to occupy almost all cpu resources (each NCK will generate an interrupt), and other tasks will not be executed

closed 2022-06-01 by rk3399