eclipse-threadx/netxduo #107: Question: How can I avoid packet pool depletion on TLS connection?

jkuwaha, 2022-06-07:
Hi,

I'm experiencing sudden connection lost on TLS when transmitting/receiving a lot of data.
I think that is caused by lack of free area in a packet pool based on the observations below.

*How can I prevent this from happening?*

Here are settings, observations, and my understandings.
Please correct me if I'm wrong.

## Settings:
- TLS 1.2 mutual authentication
- Client: NetX Duo and NetX Secure v6.1.9
  - Connect to the server and send back the received data
    - After connection is established, the client waits for data by calling nx_secure_tls_session_receive repeatedly until NX_SUCCESS is returned. NX_NO_PACKET is ignored, since it's not an error for this API.
  - NX_IP ip
    - NX_PACKET_POOL pool0 (for Rx?)
      - payload_size = 1568
      - memory_size = 32 * (1568 + sizeof(NX_PACKET))
  - NX_SECURE_TLS_SESSION tls_session
    - protocol version override: NX_SECURE_TLS_VERSION_TLS_1_2
    - packet_buffer_set: buffer_size = 17000
    - remote_certificate_allocate: raw_buffer_size = 2000
  - NX_TCP_SOCKET
    - type_of_service: NX_IP_NORMAL
    - fragment: NX_FRAGMENT_OKAY
    - time_to_live: NX_IP_TIME_TO_LIVE
    - window_size: 8192
  - NX_PACKET_POOL pool1 (for Tx)
    - payload_size = 1568
    - memory_size = 32 * (1568 + sizeof(NX_PACKET))
- Server: PC (openssl s_server, socat, etc...)
  - Send 1MB of data to the client on connect

## Observations:
- When using TCP socket instead of TLS, transferring data is completed without error.
- With TLS,
  - Human input data, slow and small amount, are transferred correctly.
  - Sending a 1MB file causes disconnection.
  - According to the server's console,
    - Authentication is ok
    - Some data are sent from the client (like 16KB, 19KB... not always the same)
  - According to captured packets,
    - Authentication is ok
    - *The client sends an alert 'Fatal(2), Internal Error(80)'*
  - By using debugger,
    - The error NX_SECURE_TLS_INVALID_STATE(103) is returned from nx_secure_tls_session_receive.
    - It seems like an error NX_NO_PACKET(1) occurred during decryption. I observed the break point at the line 836 in _nx_secure_tls_record_packet_decrypt() is hit several times (I'm not sure it's always the same line though)

## My understandings:
1. To decrypt received record, NetX Secure uses a default packet pool of the NX_IP instance.
1. Since the default packet pool is used for receiving data, the pool may have no free area, and this may cause NX_NO_PACKET(1) error internally.
1. There is no API/setting to reserve the area for decryption in the packet pool beforehand.
1. API nx_secure_tls_session_packet_buffer_set works only for certs but not for application data. (Is this an intended behavior? I had expected this works for both.)
1. When NX_NO_PACKET(1) is returned, _nx_secure_tls_session_receive_records tries to send an alert.
   1. In _nx_secure_tls_map_error_to_alert, the alert number(80) and level(2) are decided.
   1. However, allocating a packet for the alert may fail because the packet pool might be full at this point.
      1. If it fails, it doesn't send an alert at this point.
      1. _nx_secure_tls_session_reset() is called and nx_secure_tls_socket_type is set to NX_SECURE_TLS_SESSION_TYPE_NONE.
      1. Finally, _nx_secure_tls_session_receive_records returns NX_NO_PACKET(1) to the caller.
      1. The client treats NX_NO_PACKET(1) not as an error because that is defined 'No data received' for the API.
      1. The client calls nx_secure_tls_session_receive again to receive data.
      1. _nx_secure_tls_process_record is called and checks nx_secure_tls_socket_type. Since the value is NX_SECURE_TLS_SESSION_TYPE_NONE, *the function releases the packet* and returns NX_SECURE_TLS_INVALID_STATE.
      1. The function _nx_secure_tls_session_receive_records receives the value, again, tries to send an alert.
   1. After allocating the packet, _nx_secure_tls_map_error_to_alert decide the alert number(80) and level(2).
   1. As the packet is released, allocating a packet for the alert succeeds at this time.
   1. The error code NX_SECURE_TLS_INVALID_STATE is returned to the client.

If you need any further information, please let me know.

Thanks.

tstapko, 2022-06-07:
It sounds like your application isn't processing the incoming data fast enough to handle the incoming data. If the packets aren't released soon enough, the packet pool runs out of space and you'll get that error. There are a couple of reasons this could happen (FYI - I'm not on the team anymore but I designed and wrote most of the NetX TLS stack...).

I'll try address your understanding of the situation point by point to see if anything is helpful:

> 1. To decrypt received record, NetX Secure uses a default packet pool of the NX_IP instance.

Yes - the packet pool is assigned to the IP instance, and the IP instance is in turn assigned to the TCP socket that is given to TLS. Your NX_PACKET_POOL size looks to be OK to receive 1MB of data as long as the data is processed fast enough (more below). Note that NetX TLS receives fragmented **encrypted** records using the pool memory (the 1568 is the TCP packet size but TLS records are up to 16KB) . The packet_buffer_set API assigns a buffer to reconstruct a complete TLS record from TCP fragments so it can decrypt the record in one go (we could decrypt the fragments but the functionality is a tradeoff so we can support a variety of hardware encryption engines). The 17000 bytes for that buffer represents the TLS record payload max (16KB) plus a little overhead for encryption padding and the TLS header and footer.

> 2. Since the default packet pool is used for receiving data, the pool may have no free area, and this may cause NX_NO_PACKET(1) error internally.

Correct. The issue is seems to be that TLS is trying to allocate a packet from the pool but none have been released. The relevant questions are 1) where is the packet allocation call coming from? Seems like it's in _nx_secure_tls_session_receive_records so it's probably an incoming RX packet and not outgoing/TX from the app, and 2) why is the pool filling up in the first place? This is usually because the **application** isn't releasing packets back to the pool (akin to memory leaks with malloc/free) - more on this below.

> 3. There is no API/setting to reserve the area for decryption in the packet pool beforehand.

See next point.

> 4. API nx_secure_tls_session_packet_buffer_set works only for certs but not for application data. (Is this an intended behavior? I had expected this works for both.)

This is odd. The nx_secure_tls_session_packet_buffer_set API provides a working buffer to reassemble a single fragmented TLS  record from TCP fragments. It is indeed used for receiving and processing incoming X.509 certs during the TLS handshake but the packet buffer is used to reassemble/decrypt every **every** TLS record, not just certs - the remote_certificate_allocate API assigns a buffer to store the remote certificate if you want to access it after TLS is done with chain validation - the packet buffer data is overwritten by the next TLS record received during the handshake but if you assign a remote cert buffer a copy of the incoming X.509 leaf cert is placed there. If your observation is correct that might be a functional change I'm not aware of.

> 5. When NX_NO_PACKET(1) is returned, _nx_secure_tls_session_receive_records tries to send an alert...

The behavior of TLS you cover here is expected - the failed packet allocation is indeed an "internal alert" situation, but as you noted there is no packet to allocate for the outgoing alert record (this is partially a consequence of NetX using the same packet pool for RX and TX). NOTE: There's a possible bug/feature request report here to reserve enough packets (1 should be enough) in the pool to send a TLS alert in all cases. The current behavior works (packet is released and then alert is sent) but it's a bit inelegant. 

So this leads me to possible causes based on the information you provided:

1. The "fast" incoming data overwhelms NetX TCP and the packet pool is filled up. However, I'd expect NetX to start rejecting/dropping incoming packets before this point so if this is the cause then there is possibly an edge-case bug in the TCP/TLS interaction (maybe a TLS record fragment is needed but the packet for it can't be allocated/received?)
2. Something in TLS is being called with a ThreadX timeout other than NX_WAIT_FOREVER and you've hit a case where the backlog of packets isn't being processed within the timeout period and another thread (the IP stack has an internal thread) is allocating everything in the pool asynchronously. If any TLS APIs (e.g. nx_secure_tls_session_receive) are being called directly or indirectly with the "wait_option" parameter set to something other than NX_WAIT_FOREVER, try changing them to see if that's the issue.
3. Your application isn't calling nx_packet_release quickly enough. Note that TLS will only release packets that are part of the TLS handshake, used for internal TLS protocol (e.g. alerts), or packets that result in failed processing (reassembly, parsing, or decryption errors). If your application receives a packet from nx_secure_tls_session_receive, it **must** be released by the application by calling nx_packet_release in a timely fashion. If you're receiving a lot of packets, your application may simply not be processing the data fast enough to release packets to keep space in the packet pool.
4. Your application is using the same TCP socket to send data other than TLS records. Since NetX TLS decouples the TCP socket from TLS (unlike OpenSSL, for example) it is technically possible to send non-TLS data out o [... cut ...]
I'd be curious to see where the failed allocation is coming from - if this is indeed an edge case, it would be a case where TLS is trying to allocate a packet for something that hasn't been accounted for - it might be that the allocation is too large for a single packet and it tries to chain the packet (see NetX packet chaining) but TLS only accounted for a single packet and nx_packet_allocate doesn't know any better.

Anyway, hopefully some of this is helpful. The Azure RTOS team also monitors this forum so they might be able to help as well.

jkuwaha, 2022-06-08:
Thank you very much, tstapko!

Your detailed and insightful comments are very helpful.
I think it'll take a little time for me to digest them all because network is not my specialty.

For now, I'm leaving comments that I can write right away.

> This is odd. The nx_secure_tls_session_packet_buffer_set API provides a working buffer to reassemble a single fragmented TLS record from TCP fragments. It is indeed used for receiving and processing [...]

Regarding this, I shouldn't have mentioned 'certs'. I just wanted to say 'isn't used for application data' while the documentation implies to do so.

Following is the reason why I think so.

`_nx_secure_tls_session_packet_buffer_set` sets the member `nx_secure_tls_packet_buffer` of the NX_SECURE_TLS_SESSION and is an only setter for the member.
The same member is accessed from `_nx_secure_tls_process_record` and `_nx_secure_tls_process_remote_certificate`. I don't dig deeper the latter.

Here is an excerpt form `_nx_secure_tls_process_record`. It seems like only when message type is **not** NX_SECURE_TLS_APPLICATION_DATA, `nx_secure_tls_packet_buffer` is used. And this is the only place where `nx_secure_tls_packet_buffer` is used from `_nx_secure_tls_process_record`.
Of course, the value is copied to `packet_data` which can be used outside of this `if` clause, considering copy is made only when message type is **not** NX_SECURE_TLS_APPLICATION_DATA, there is no need to check the use of that variable.

https://github.com/azure-rtos/netxduo/blob/1eee883def078a91b8cd19159d1796e3a8862814/nx_secure/src/nx_secure_tls_process_record.c#L377-L401

On the other hand, decrypting application data is happening earlier in the same function.

https://github.com/azure-rtos/netxduo/blob/1eee883def078a91b8cd19159d1796e3a8862814/nx_secure/src/nx_secure_tls_process_record.c#L243-L275

The function `_nx_secure_tls_record_payload_decrypt` calls the function `_nx_secure_tls_record_chained_packet_decrypt` where new packet allocation may be made.

https://github.com/azure-rtos/netxduo/blob/1eee883def078a91b8cd19159d1796e3a8862814/nx_secure/src/nx_secure_tls_record_payload_decrypt.c#L607-L626

And the function `_nx_secure_tls_record_packet_decrypt` called from the above excerpt may allocate another packet.

https://github.com/azure-rtos/netxduo/blob/1eee883def078a91b8cd19159d1796e3a8862814/nx_secure/src/nx_secure_tls_record_payload_decrypt.c#L828-L835

Considering the above, I concluded that API `nx_secure_tls_session_packet_buffer_set` doesn't work for application data.

> I'd be curious to see where the failed allocation is coming from - if this is indeed an edge case, it would be a case where TLS is trying to allocate a packet for something that hasn't been [...]

For the first part,
I placed two breakpoints at line 604 and 833 in nx_secure_tls_record_payload_decrypt.c, 
at least one of them was hit.
I'm sorry, I don't remember exact which.

I'm afraid I couldn't catch your point of the latter part.
Are you mentioning something concrete?

Regarding decrypting application data, it seems fine. The function `_nx_secure_tls_record_chained_packet_decrypt` allocates the first packet of the decrypted data, and calls repeatedly `_nx_secure_tls_record_packet_decrypt` that decrypts as much data as the packet can have and allocates new packet if there is no room in the packet.

I guess perhaps providing a dedicated buffer/packet pool for decryption can solve the problem.
What do you think?

Again, I deeply appreciate for your help, tstapko.

TiejunMS, 2022-06-08:
The TLS session will use IP packet pool (set in `nx_ip_create`) to perform decryption. Could you try to increase the packet pool size and see if the problem can be solved?

tstapko, 2022-06-08:
> Thank you very much, tstapko!
>
> Your detailed and insightful comments are very helpful. I think it'll take a little time for me to digest them all because network is not my specialty.
>
> For now, I'm [...]

Ah, yes. That is a later modification to the stack that I forgot about (and didn't write myself) - the ability to decrypt fragments was added later and decrypts application data in the packet pool directly. This is an optimization that eliminates the need to copy data into and out of the packet buffer (improving throughput) so technically you are 100% correct - the TLS packet buffer is now only really used for certificates during the handshake (parsing X.509 fragments would be technically difficult without much gain). 

> For the first part, I placed two breakpoints at line 604 and 833 in nx_secure_tls_record_payload_decrypt.c, at least one of them was hit. I'm sorry, I don't remember exact which.
>
> I'm afraid I [...]

NetX packet allocation is configurable so that it can automatically handle "chaining" packets together - if you allocate packet space that is larger than the configured packet size, NetX will create a linked list ("chain") of packets, automatically fragmenting the data for you. I'm not an expert on how/when NetX utilizes chaining @TiejunMS is the expert there. 

My point was that TLS might be saving a single packet in the pool for alerts but a previous allocation automatically grabbed multiple packets - an edge case that TLS might not account for, leading to the odd behavior of trying to send an alert, failing, releasing a packet, and trying again. I don't think it's a serious issue, but it's probably worth investigating.

> Regarding decrypting application data, it seems fine. The function `_nx_secure_tls_record_chained_packet_decrypt` allocates the first packet of the decrypted data, and calls repeatedly [...]

I think your best bet is as @TiejunMS indicated - increasing the packet pool size can possibly resolve the issue. Even if that doesn't work, the fact it doesn't gives good insight into what actually may be happening. Good luck!

jkuwaha, 2022-06-14:
Thank you, @tstapko and @TiejunMS

I made some experiments with different packet pool sizes.
During the experiments I found a misconfiguration of my network and fixed it.
After that, all tests passed using **socat** command,
however, tests still failed with openssl s_server.

## Experiment
The server (PC) transfers 1MiB of data using `cat test_data | openssl s_server`.
The client (NetX duo) connects to server and sends all the data received from the server back to the server.

I made experiments changing size of packet pools (both NX IP's and sender's socket).
Also, I made a small change to `_nx_secure_tls_session_start` to use a dedicated packet pool for TLS as follows;
```
    /* Assign the packet pool from which TLS will allocate internal message packets. */
    extern NX_PACKET_POOL *pp4tls;
    tls_session -> nx_secure_tls_packet_pool = pp4tls ? pp4tls : tcp_socket -> nx_tcp_socket_ip_ptr -> nx_ip_default_packet_pool;
```

## Results

Size of packet pool | Result | Error at
----|----|----
32 x (1568 + sizeof(NX_PACKET)) | NX_SECURE_TLS_INVALID_STATE | receive
64 x (1568 + sizeof(NX_PACKET)) | NX_WINDOW_OVERFLOW | send
128 x (1568 + sizeof(NX_PACKET)) | NX_WINDOW_OVERFLOW | send
32 x (1568 + sizeof(NX_PACKET)) + 16KB pool | NX_WINDOW_OVERFLOW | send

For 128x setting, if I modify the client to send only, I confirm all data are handled.

## Regarding 'packet pool depletion' problem

As you said, TLS_INVALID_STATE is gone if size of the pool is increased.

However, I still feel like this couldn't be a permanent solution.
Here's why.

NX_IP's pool is used for incoming packets. TCP window prevents the pool from depletion. (am I right?)
TLS also uses NX_IP's pool when an application calls 'nx_secure_tls_session_receive' from the application's thread.
So, under heavy network load, IP's pool might be already full when 'nx_secure_tls_session_receive' is called,
or TLS may use the last packet in the pool, which may cause NX_IP fail to allocate for incoming packets or even ACK.

I think using dedicated packet pool during TLS processing could be a solution.
I found `tls_session` has a member `nx_secure_tls_packet_pool`, and it seems like code for dedicated pool is ready.
Why wasn't this used? Are there any technical difficulties?

## Regarding 'window overflow' problem

When is NX_WINDOW_OVERFLOW returned? What should I do?
Is it an application's responsibility to assure sending size is less than the current window size?

[image]

According to packet capture, it seems like the server (10.72.245.160) didn't send ACK to the client (10.72.245.29).
Is this okay?
Also, are followings okay?
1. At #1456, the client sent a packet with length 1 while the server reports its window is 0.
2. At #1457, the server sent an ACK for 648716 not for 648717.
3. At #1458, #1460 and #1462, the client sent a zero length packet but 'next sequence number' is incremented by 1.

If you need any additional information, please let me know.

tstapko, 2022-06-14:
I'm not an expert in TCP subtleties (@TiejunMS would be able to help a lot more here) but I would guess that the application does need to be aware of the TCP window size to some extent. If the application is trying to send data but the pool is full (probably because the physical network layer is congested?) then I'd expect errors when trying to send data. Given your test case (receiving data and echoing it back) I would expect issues if the application turns the data around immediately and sends it back out on the same TLS session and the physical connection (e.g. WiFi adapter on your RTOS device) is slow. 

One experiment to try is to make the packet pool large enough to hold twice the data being sent by OpenSSL (plus some extra). If your device has enough RAM, make the packet pool something like 2.1MB. If you don't have the RAM, reduce the size of data being sent. I'd expect that if the packet pool can hold all of the incoming and outgoing data simultaneously, you shouldn't see any problems. 

If you reduced the data being sent, then try to reduce the packet pool size to something smaller and see if you observe the same behavior (though you might not see issues with a smaller total data load). Also try just receiving the data and not sending anything back. Maybe do some calculation on the data (e.g. a checksum) before releasing the incoming packets to give more of a "real-world" setup. My guess is that you're simply outpacing the hardware's ability to send and receive, leading to a backup which appears as a packet pool issue. The fact that increasing the pool size changed the symptoms seems to point in that direction.

As for the nx_secure_tls_packet_pool member, that should just be an internal-use reference to the IP packet pool and that would only be used for incoming data and TLS protocol messages (handshake, alerts). Your change to the TLS API might cause some issues since you'd now be using separate packet pools for TX and RX and those pools are not necessarily properly managed by the IP instance and TCP socket (this is something @TiejunMS would have to clarify as it's not a NetX use-case I'm familiar with).

jkuwaha, 2022-06-16:
Thank you, @tstapko and @TiejunMS

I'll do some additional experiments and get back to you.

To @tstapko,

> Given your test case (receiving data and echoing it back) I would expect issues if the application turns the data around immediately and sends it back out on the same TLS session and the physical connection (e.g. WiFi adapter on your RTOS device) is slow.

Thank you for your advice.
Maybe it is slow and has limited resources. However, I want TLS is reliable under the such condition. (I don't know if it's possible, though.) Is it impossible by nature of TLS? If not, it would be nice it's provided.

To @TiejunMS,

> We will consider adding an API to support separate packet pool for TLS session.

Thank you very much for considering that. I hope that will solve my problem.

> TLS is sensitive to any transmission failure from TCP layer. If the wait option of `nx_secure_tls_session_send` is set to `NX_WAIT_FOREVER`, then user application does not need to take care of the TCP window size. Or else, it is possible `nx_tcp_socket_send` will return failure which causes a reset at TLS layer. What is the wait option in your case?

Thank you for your explanation. For now, wait option is set to 1 sec, so I'll change it to NX_WAIT_FOREVER and do some experiments.

jkuwaha, 2022-06-20:
I have made experiments.

Firstly, I limited the data size to 128KiB where the size of the packet pool is around 128KiB.
I repeated the test 10 times, and data are transferred correctly every time.

Secondly, I set WAIT_FOREVER to the timeout of nx_secure_tls_session_send, I tried to transfer 1MiB of data.
WINDOW_OVERFLOW didn't occur, but transfer didn't completed. According to the packet capture, both windows were 0 at the server and client, seems like dead locked.

> Also review the specs of your network hardware (WiFi or Ethernet PHY?) to make sure your application isn't doing more than the hardware can handle.

Thank you, @tstapko.
I still feel like flow control might work for this kind of situation, but I also understand your point. I'll consult with the chip manufacturer for better setting.

TiejunMS, 2022-06-20:
@jkuwaha , could you share the packet trace for the dead lock issue? 
Since one end is PC, the TCP window size should be able to recover unless data is not received by application. What is the exact command you used for testing?

jkuwaha, 2022-06-21:
@TiejunMS,
I made the same experiment, and PC to PC transfer experiment for reference.
I can't put the whole pcap file to the public places for some reasons, so if you need them, please let me know with your contact information.

## Experiment

The server sends 1MiB of data to the client and the client sends back to the server; where the client is NetX.

### Setting

Server: OpenSSL 1.1.1f 31 Mar 2020 on Ubuntu 20.04.4 LTS on WSL1

```sh
(head -c 1024KiB test.txt; cat Q; sleep 3) | openssl s_server -port 9999 -quiet -cert s5.crt -key s5.key -CApath . -Verify 10 -verify_return_error -keylogfile keylog
```

where test.txt contains incrementing counters one per line in hex ('|%06X\n'), s5.crt is a certificate, and s5.key is a private key for the server.
(`cat Q` and `sleep 3` might not be necessary.)

### Result
Please check the screenshot below. 10.72.245.160 is the server, 10.72.245.36 is the client (NetX).

[image]

At #1421 and #1422, both sides state window size 0. And repeat the same forever.

## Reference

The server sends 1MiB of data to the client.
Instead of sending back what the client receives, the client sends 1MiB of data from its own data source.
Please note that the setting is not exactly same as the experiment on NetX.
Both the server and the client are on Ubuntu on PC.

### Setting

Server: Same as above

Client: OpenSSL 1.1.1f 31 Mar 2020 on Ubuntu 20.04.1 on VirtualBox

(note: server and client are hosted by physically different machines)

```sh
(head -c 1024KiB test.txt; cat Q; sleep 3) | openssl s_client -connect 10.72.245.160 -port 9999 -servername s5 -cert c3.crt -build_chain -key c3.key -CApath . -chainCApath . -strict -debug -showcerts -verify 10 -verify_return_error -tls1_2
```

### Result
Please check the screenshot below. 10.72.245.160 is the server, 10.72.245.20 is the client.

[image]

## Comment

> Since one end is PC, the TCP window size should be able to recover unless data is not received by application. What is the exact command you used for testing?

I think the server's behavior is strange. I suspected it's because of WSL and made an experiment with a server running on Ubuntu on VirtualBox, but the result was the same.

By the way, if the same can be applied to the client, does this imply there is something wrong at the client side? Is there anything I can do with my code?

TiejunMS, 2022-06-21:
@jkuwaha , could you share the packet trace and test.txt by email? Here is the email address: azure-rtos-support@microsoft.com. Thanks! I will try to reproduce the issue on my side.

TiejunMS, 2022-06-21:
@jkuwaha , I have created a simulation project [here](https://github.com/TiejunMS/netxduo_projects) but I can not reproduce your issue. [Here](https://github.com/TiejunMS/netxduo_projects/blob/main/issue_107/README.MD) is a simple readme on how to run simulation project. Could you take a look at my [logic ](https://github.com/TiejunMS/netxduo_projects/blob/1d7d4e5d544c451ac54ff6a43b515958bfeed2b0/issue_107/main.c#L256)of receiving and sending data and see if it differs from yours?

jkuwaha, 2022-06-22:
@TiejunMS,
Thank you. 
I'll port your code to my environment, see if reproduce, and get back to you.

>  Could you take a look at my logic of receiving and sending data and see if it differs from yours?

I noticed that buffering strategy is different from mine.
In my code, received data are copied from a `NX_PACKET` as per data consumer's request. If the packet size is larger than the request, the packet is used for several requests until all data are used. Here's a pseud-code.

```c

NX_PACKET *receive_packet = NULL;
uint32_t next_pos = 0;

static size_t my_receive(NX_SECURE_TLS_SESSION *session_ptr, uint8_t *buffer, size_t size)
{
    while (true) {
        // Read form packet
        if (receive_packet) {
            // 1. copy data from already received packet to buffer
            // 2. update next_pos
            // 3. if all data in received packet are copied, release the packet and set NULL to receive_packet.
            // 4. return if buffer is fully filled, timeout or error)
        }
        // Receive
        //  - Receive data into packet by using 'nx_secure_tls_session_receive()'
    }
}

static size_t my_send(NX_SECURE_TLS_SESSION *session_ptr, uint8_t *buffer, size_t size)
{
    while (byte_sent < size) {
        // 1. Allocate packet by 'nx_secure_tls_packet_allocate()'
        // 2. Copy data form buffer to packet by 'nx_packet_data_append()'
        // 3. Send data by 'nx_secure_tls_session_send()'.
        // 4. If 'nx_secure_tls_session_send()' returns error, release the packet and return;
    }
    return byte_sent;
}

static void echo(NX_SECURE_TLS_SESSION *session_ptr)
{
    static uint8_t buffer[1024];
    while (true) {
        size_t received = my_receive(session_ptr, buffer, sizeof(buffer));
        my_send(session_ptr, buffer, received);
    }
}

```

Is there something wrong with my logic?

TiejunMS, 2022-06-22:
Your logic looks good. If you are not using `nx_packet_data_extract_offset` to retrieve data, I would suggest you to this API rather than adjusting the packet structure in user application.

jkuwaha, 2022-06-22:
I ported your logic and it runs without error; Not reproduced even in my environment! I'll change your logic to mine gradually and see what happens.

> Your logic looks good. If you are not using nx_packet_data_extract_offset to retrieve data, I would suggest you to this API rather than adjusting the packet structure in user application.

Thank you for your comment. And yes, I use 'nx_packet_data_extract_offset' to extract data.
By the way, is it possible to know how much data can be added to the packet without internal new allocation? If possible, I can call 'nx_packet_data_append` with NO_WAIT safely.

TiejunMS, 2022-06-22:
> I ported your logic and it runs without error; Not reproduced even in my environment! I'll change your logic to mine gradually and see what happens.

Let me know what you find. 

> By the way, is it possible to know how much data can be added to the packet without internal new allocation? If possible, I can call 'nx_packet_data_append` with NO_WAIT safely.

It is very hard to determine the size of data that can be transmitted at any moment. See the logic between line [445 ](https://github.com/azure-rtos/netxduo/blob/93aff552c9e85c3b96f7ae62d0cdf48649d641ea/common/src/nx_tcp_socket_send_internal.c#L445)and 483. The two major factors are RX window (from peer) and congestion window. Those two values are both related the peer. It is possible those two values can be reduced to zero at any moment. Unless the user application locks the IP mutex during transmission, or else the NO_WAIT is not a safe timeout for TLS transmission. But there is also a risk of mutex dead lock for user application to lock the IP mutex.

However, you can still use this API `nx_tcp_socket_info_get` to get `tcp_transmit_window` and set packet size to avoid hitting window overflow for best effort. Note, you will need to reduce the size for TLS header, IV, ICV and padding as well.

jkuwaha, 2022-06-22:
With my logic and reduced buffer, it reproduced. What I modified from your code is:

- buffering strategy has been changed
- variable `packet_ptr_s` has been introduced for send
- varialbe `receive_size_max` has been introduced for limit the size of buffer
- (type of pool_0 has been changed to pointer to fit my environment.)

I tested 3 times changing `receive_size_max`.

`receive_size_max` | result
:----:|:----:
1024 | 3/3 Reproduced
2048 | 3/3 Reproduced
4096 | 3/3 Reproduced
8192 | 1/3 reproduced
17000 | Not reproduced

I'll check and analyze the results. Could you please try the same on your environment?

```c
NX_PACKET *packet_ptr_s = NULL;
NX_PACKET_POOL *pool_0 = &packetPool;
ULONG receive_size_max = BUFFER_SIZE;

for (;;)
{
    /* Wait for a packet.  */
    if (!packet_ptr) {
        status = nx_secure_tls_session_receive(&tls_client_session_0, &packet_ptr, NX_WAIT_FOREVER);
        if (status)
        {
            ERROR_COUNTER();
            break;
        }
        total_bytes += packet_ptr -> nx_packet_length;
        printf("Received packet: %05lu bytes, total %07u bytes", packet_ptr -> nx_packet_length, total_bytes);

        pos_in_packet = 0;
    }

    /* Extract the data from the packet.  */
    ULONG remain = packet_ptr->nx_packet_length - pos_in_packet;
    ULONG length = (remain < receive_size_max) ? remain : receive_size_max;
    status = nx_packet_data_extract_offset(packet_ptr, pos_in_packet, receive_buffer,
                                            length, &bytes_copied);
    if (status)
    {
        ERROR_COUNTER();
        nx_packet_release(packet_ptr);
        break;
    }
    pos_in_packet += bytes_copied;
    printf("Data copied: %05u bytes, pos in packet: %05u", bytes_copied, pos_in_packet);

    /* Release received packet.  */
    if (packet_ptr->nx_packet_length <= pos_in_packet) {
        nx_packet_release(packet_ptr);
        packet_ptr = NULL;
        pos_in_packet = 0;
        printf("Packet released (pos in packet %05u)", pos_in_packet);
    }

    /* Allocate a TLS packet for transmission.  */
    status = nx_secure_tls_packet_allocate(&tls_client_session_0, pool_0,
                                            &packet_ptr_s, NX_WAIT_FOREVER);
    printf("Packet allocated");
    if (status)
    {
        ERROR_COUNTER();
        break;
    }

    /* Append data to the packet.  */
    status = nx_packet_data_append(packet_ptr_s, receive_buffer, bytes_copied, pool_0,
                                    NX_WAIT_FOREVER);
    printf("Data appended %05u", bytes_copied);
    if (status)
    {
        ERROR_COUNTER();
        nx_packet_release(packet_ptr_s);
        break;
    }

    /* Send the data back to the server.  */
    status = nx_secure_tls_session_send(&tls_client_session_0, packet_ptr_s, NX_WAIT_FOREVER);
    printf("Data sent");
    if (status)
    {
        ERROR_COUNTER();
        nx_packet_release(packet_ptr_s);
        break;
    }
}
```

TiejunMS, 2022-06-22:
I can reproduce the issue with your logic. I took a look at the [openssl s_server](https://github.com/openssl/openssl/blob/master/apps/s_server.c) and I think the issue is in the application logic but not TCP or TLS layer. Both NetX and openssl s_server **applications** are not receiving data **in time** which cause the TCP RX window of both sides dropped to zero. Here are the steps how this issue occur.
1. NetX connects to openssl s_server.
2. NetX waits for data from openssl.
3. openssl [sends ](https://github.com/openssl/openssl/blob/bcb52bcc9f9c36a85d037976676fd5ca52f307cd/apps/s_server.c#L2704)16K data to NetX.
4. NetX sends 1K (2K/4K/8K/17K) data to openssl.
5. openssl [receives ](https://github.com/openssl/openssl/blob/bcb52bcc9f9c36a85d037976676fd5ca52f307cd/apps/s_server.c#L2790)data from NetX.
6. Goto step 3 when NetX receive buffer is not empty. This won't happen when the `receive_size_max` is 17K.
7. Goto step 2 when all received data from NetX to openssl are sent.

In the loop between step 3 and 6, NetX TCP RX window will drop at each round till zero. Then openssl will not be able to send any data and stuck at step 3. And openssl is not able to receive any data either. Finally, NetX will not be able to send data due to openssl TCP RX window is also dropped to zero. Both sides are stuck at their send routine.

Possible solution to solve it is to split the logic of sending and receiving for both sides.

jkuwaha, 2022-06-23:
Thank you very much for your comment, @TiejunMS.
Now I can understand what was happening.
And I should use a multi-threaded application as a server. I feel sorry for bothering you.

Also, thank you to answering my question.

> It is very hard to determine the size of data that can be transmitted at any moment.

I understood. I'll leave the complex window handling to NetX.

I'd like to wrap up this issue before close. Please correct me if I'm wrong. If it seems okay to you, please close the issue.

1. `NX_SECURE_TLS_INVALID_STATE` reported under low memory situation
  - NetX dev team to consider adding an API to support separate packet pool for TLS session.
  - What users can do:
    - to increase the size of user application's packet pool to lower its possibility.
    - to seek chip manufacturer's advice.

2. `NX_WINDOW_OVERFLOW` reported during echo test with OpenSSL s_server
  - Not a issue. NetX works as expected.
    - `nx_secure_tls_session_send` may return `NX_WINDOW_OVERFLOW` on timeout.
    - Structure (single thread receive and send) of the user application and/or s_server is not suitable for echo test.

Again, I really appreciate @tstapko and @TiejunMs. Thank you.

TiejunMS, 2022-06-23:
@jkuwaha , you are very welcome! I'm so glad your issue is solved. And thanks for the wrapping up! It will be helpful to users who have similar issue. Closing.

closed 2022-06-23 by TiejunMS