eclipse-threadx/netxduo #113: Nx_web_http_server_secure_configure() improper remote certificate list management

leobbditestcom, 2022-07-06:
When there is more than one remote certificate being allocated, nx_web_http_server_secure_configure() attempts to re-add the same certificate to remote certificates list, because certificates are being indexed by session index instead of certificate index.  This problem went away when the following change was made to nx_tcpserver.c, line 533:

                //status = nx_secure_tls_remote_certificate_allocate(tls_session, remote_certificates[i],
                status = nx_secure_tls_remote_certificate_allocate(tls_session, remote_certificates[cert_count],

wenhui-xie, 2022-07-07:
@leobbditestcom Thanks for reporting this issue, we will fix it in the next release.

bo-ms, 2022-08-01:
@leobbditestcom closing, the bug fix is released in 6.1.12 now.

closed 2022-08-01 by bo-ms